analogdevicesinc/iio-oscilloscope #195: Deprecated usage of GTimeVal and GTypeDebugFlags

JulioSDDR, 2019-10-27:
Error output:

[  1%] Building C object CMakeFiles/osc.dir/osc.c.o
In file included from /usr/include/gtk-2.0/gtk/gtkobject.h:37,
                 from /usr/include/gtk-2.0/gtk/gtkwidget.h:36,
                 from /usr/include/gtk-2.0/gtk/gtkcontainer.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkbin.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkwindow.h:36,
                 from /usr/include/gtk-2.0/gtk/gtkdialog.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkaboutdialog.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:33,
                 from /home/julio/.cache/yay/iio-oscilloscope-git/src/iio-oscilloscope/osc.c:7:
/usr/include/gtk-2.0/gtk/gtktypeutils.h:236:1: error: ‘GTypeDebugFlags’ is deprecated [-Werror=deprecated-declarations]
  236 | void            gtk_type_init   (GTypeDebugFlags    debug_flags);
      | ^~~~
In file included from /usr/include/glib-2.0/gobject/gobject.h:24,
                 from /usr/include/glib-2.0/gobject/gbinding.h:29,
                 from /usr/include/glib-2.0/glib-object.h:23,
                 from /usr/include/glib-2.0/gio/gioenums.h:28,
                 from /usr/include/glib-2.0/gio/giotypes.h:28,
                 from /usr/include/glib-2.0/gio/gio.h:26,
                 from /usr/include/gtk-2.0/gdk/gdkapplaunchcontext.h:30,
                 from /usr/include/gtk-2.0/gdk/gdk.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:32,
                 from /home/julio/.cache/yay/iio-oscilloscope-git/src/iio-oscilloscope/osc.c:7:
/usr/include/glib-2.0/gobject/gtype.h:679:1: note: declared here
  679 | {
      | ^
In file included from /usr/include/gtk-2.0/gtk/gtktoolitem.h:31,
                 from /usr/include/gtk-2.0/gtk/gtktoolbutton.h:30,
                 from /usr/include/gtk-2.0/gtk/gtkmenutoolbutton.h:30,
                 from /usr/include/gtk-2.0/gtk/gtk.h:126,
                 from /home/julio/.cache/yay/iio-oscilloscope-git/src/iio-oscilloscope/osc.c:7:
/usr/include/gtk-2.0/gtk/gtktooltips.h:73:3: error: ‘GTimeVal’ is deprecated: Use 'GDateTime' instead [-Werror=deprecated-declarations]
   73 |   GTimeVal last_popdown;
      |   ^~~~~~~~
In file included from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/glib-2.0/gobject/gbinding.h:28,
                 from /usr/include/glib-2.0/glib-object.h:23,
                 from /usr/include/glib-2.0/gio/gioenums.h:28,
                 from /usr/include/glib-2.0/gio/giotypes.h:28,
                 from /usr/include/glib-2.0/gio/gio.h:26,
                 from /usr/include/gtk-2.0/gdk/gdkapplaunchcontext.h:30,
                 from /usr/include/gtk-2.0/gdk/gdk.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:32,
                 from /home/julio/.cache/yay/iio-oscilloscope-git/src/iio-oscilloscope/osc.c:7:
/usr/include/glib-2.0/glib/gtypes.h:551:8: note: declared here
  551 | struct _GTimeVal
      |        ^~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/osc.dir/build.make:63: CMakeFiles/osc.dir/osc.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/osc.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...

dNechita, 2019-10-28:
Hi,
We're working on a solution.
Thanks for reporting this!

dNechita, 2019-10-29:
A fix has been made. Checkout the latest master.

closed 2019-11-05 by dNechita