analogdevicesinc/iio-oscilloscope #199: Can't open settings window correctly

DynamixYANG, 2019-11-13:
The iio app application seems to recognize the pluto, but the settings page doesn't show up. The capture page is working fine, and the waveform is shown smoothly. This issue still occurs after I replug the USB port and reinstall the driver and iio app. I use the latest iio osc version and driver.

dNechita, 2019-11-13:
What operating system are you using?

DynamixYANG, 2019-11-14:
Windows 10 1903. By the way, the settings window seems running but can't properly display. There is a task logo in the taskbar but show up a shadow when I click it.

dNechita, 2019-11-15:
If you click on that logo from the taskbar and then use the windows button in combination with the arrow buttons, can you bring the window to the visible area? For example: hold the windows button and press left arrow (once or twice) or the right arrow.

DynamixYANG, 2019-11-19:
This issue doesn't appear after the second time I re-install the pluto driver and oscilloscope app. I haven't tried the method you mentioned above. I will tell you when it reappears.

DynamixYANG, 2019-11-28:
This issue appears again. I tried your way above, but the settings window still can't properly display. I connect the pluto to another Win10 laptop and the settings page can display properly after connecting the device. I try to uninstall the iiolib, osc app, plutoM2K driver and reinstall them for several times, but it seems no affect.

dNechita, 2019-11-28:
Are you using a multiple displays(monitors)?

DynamixYANG, 2019-11-29:
No, only on my laptop. Are there any user pofiles for osc app in another place? I want to delete it.  Because every time after I re-install the osc app, it can automaticly connect my pluto, pop up the capture window and the settings window just flash and disappear.

dNechita, 2019-11-29:
On Windows, the user profile should be at:
C:\Users\your_user_name\AppData\Local\.osc_profile.ini

spectrum05, 2020-02-29:
Hello, I have exactly the same problem as DynamixYang with windows 8.1 and Pluto; I could not resolve it even after reinstalling many times , I have a dual display but tried also with only one screen.
Pluto is well recognized but the settings window does not appear although the display window is here. Very strange. Any Idea ?

tfcollins, 2020-02-29:
What language is your OS set to?

spectrum05, 2020-03-01:
*French*
*Could it be the reason?*

tfcollins, 2020-03-02:
That is likely the issue. Different locales tend to break IIO-Scope.

spectrum05, 2020-03-02:
*However, I just tried to a Lenovo laptop with the same windows 8.1 french
version and it works fine.*

*I don't understand what is going on with my fixed PC.*

rgetz, 2020-06-26:
I had a quick look at some of the source - error conditions for window placements aren't handed correctly.

I can't reproduce it, but I have seen it, and I can see logically why it might happen.

dNechita, 2020-07-27:
This has been fixed. If the error persists, please feel free to re-open this issue.

closed 2020-07-27 by dNechita

spectrum05, 2020-07-27:
*I cannot find your latest version V0.12 for windows ? V0.11 does not open
correctly still on my computer.*
*Regards*

*Alain Azoulay,  *

*Fontenay aux Roses, France*

tfcollins, 2020-07-27:
There is no v0.12 yet. If you want to test Windows you can grab a beta installer from the CI server (See artifacts of a given build): https://ci.appveyor.com/project/analogdevicesinc/iio-oscilloscope/history

-Travis

dNechita, 2020-08-03:
> *I cannot find your latest version V0.12 for windows ? V0.11 does not open correctly still on my computer.* *Regards* *Alain Azoulay, * *Fontenay aux Roses, France* Le lun. 27 juil. 2020 à 10:51, Dan Nechita <notifications@github.com> a écrit :
> […](#)

Hi,

We have a release candidate for v0.12. If you want to check that your issue has been fixed, you can find the release here: https://github.com/analogdevicesinc/iio-oscilloscope/releases/tag/v0.12-rc.1 along with a windows installer.

-Dan

spectrum05, 2020-08-03:
Hi,
Thank you for your help, V0.12 is OK and works fine on my PCs (desktop and
laptop)
Regards
Alain

*Alain Azoulay,  *

*Fontenay aux Roses, France*

dNechita, 2020-08-04:
Thank you for the feedback!

ferdem, 2023-11-23:
IIO Oscilloscope was showing a pop-up saying a new version is available. Finally, I updated it and the same problem mentioned in this issue appeared. I tried to install older versions (without un-installing existing setup) but it did not work. Then, I realized that it is required to uninstall the existing IIO Oscilloscope from the control panel and clean-up the installation directory and make a fresh install. Finally, it works fine.