analogdevicesinc/iio-oscilloscope #210: DAQ2: Can not change the attributes of the second TX channel

Csomi, 2019-12-10:
Can not set the frequency, scale or phase attributes for the second TX channel. This behavior exists in all the DDS modes, except 'DAC Buffer'.

mhennerich, 2020-08-03:
That's due to the DAC data manager thinking the two channels are complex. So selecting independent IQ mode, should allow you controlling the second channel frequency, scale and phase. Would it be possible to add a checkbox to the DAC data manager to control it's behavior?  Any other idea? The issue is actually on the kernel side. For all new devices we use I,Q modified channels. However when we first started, those modifiers did not exist. AD9361 for example is IQ but uses unmodified channels. We can't fix this without breaking user space....

dNechita, 2020-08-03:
It is possible to add a checkbox to the DAC data manager to switch between current mode (I-Q mode) and a new mode (non I-Q) but it is not a quick fix, it will require quite some work.