analogdevicesinc/iio-oscilloscope #212: Fails to compile on Ubuntu 18

MikeLooijmans, 2019-12-16:
`/home/mike/projects/iio-oscilloscope/plugins/spectrum_analyzer.c:639:8: error: implicit declaration of function ‘ad9361_set_trx_fir_enable’ [-Werror=implicit-function-declaration]
  ret = ad9361_set_trx_fir_enable(dev, true);
        ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/mike/projects/iio-oscilloscope/plugins/fmcomms2.c: In function ‘tx_sample_frequency_changed_cb’:
/home/mike/projects/iio-oscilloscope/plugins/fmcomms2.c:374:3: error: implicit declaration of function ‘ad9361_set_bb_rate’ [-Werror=implicit-function-declaration]
   ad9361_set_bb_rate (dev, (unsigned long) (rate * 1000000));
   ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
plugins/CMakeFiles/spectrum_analyzer.dir/build.make:62: recipe for target 'plugins/CMakeFiles/spectrum_analyzer.dir/spectrum_analyzer.c.o' failed
make[2]: *** [plugins/CMakeFiles/spectrum_analyzer.dir/spectrum_analyzer.c.o] Error 1
CMakeFiles/Makefile2:281: recipe for target 'plugins/CMakeFiles/spectrum_analyzer.dir/all' failed
make[1]: *** [plugins/CMakeFiles/spectrum_analyzer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/mike/projects/iio-oscilloscope/plugins/fmcomms2.c: In function ‘filter_fir_update’:
/home/mike/projects/iio-oscilloscope/plugins/fmcomms2.c:658:2: error: implicit declaration of function ‘ad9361_get_trx_fir_enable’ [-Werror=implicit-function-declaration]
  ad9361_get_trx_fir_enable(dev, &stat);
  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/mike/projects/iio-oscilloscope/plugins/fmcomms2.c: In function ‘filter_fir_enable’:
/home/mike/projects/iio-oscilloscope/plugins/fmcomms2.c:691:3: error: implicit declaration of function ‘ad9361_set_trx_fir_enable’; did you mean ‘filter_fir_enable’? [-Werror=implicit-function-declaration]
   ad9361_set_trx_fir_enable(dev, rxtx);
   ^~~~~~~~~~~~~~~~~~~~~~~~~
   filter_fir_enable
`

dNechita, 2019-12-16:
Seems like libad9361 is not installed on your system. Did you build and install https://github.com/analogdevicesinc/libad9361-iio?

MikeLooijmans, 2019-12-16:
Obviously not. It isn't mentioned in the README and the CMakeLists.txt also doesn't refer it.

dNechita, 2019-12-16:
Sorry about that.
We keep most of the doc here, at: https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope
Please let me know if you have any further issues.

MikeLooijmans, 2019-12-16:
Compiles and works okay now. 

Guess the real issue is that CMake's "find_library" just returns a NOTFOUND value when the lib isn't available. The CMakeLists.txt apparently needs an extra if()... following the call to inform the user that the library is not there.

closed 2019-12-16 by MikeLooijmans