analogdevicesinc/iio-oscilloscope #240: CN0508 plugin bug/enhancement

mthoren-adi, 2020-05-24:
@mirceacaprioru looks like the CN0508 plugin doesn't set the AD7124-4 sample rate to 9600sps. It hangs otherwise due to the 10sps default sample rate. (Still need to run that separate script before running Osc.)

Also an enhancement request - could you change the DAC output field to "Output Voltage", where a user input of 0-27.5 (with limits enforced) corresponds to a DAC code of 0-65535. The present "persistence" behavior is desirable, that is, the number entered doesn't reset to zero (the AD5683 is write-only, and the raw attribute always reads zero).
A default value of 12.0V on running would be fine - the user can set the voltage pot lower if needed.

mirceacaprioru, 2020-05-24:
Will do some development on it tomorrow.

mirceacaprioru, 2020-06-09:
Solved with https://github.com/analogdevicesinc/iio-oscilloscope/pull/241

closed 2020-06-09 by mirceacaprioru