analogdevicesinc/iio-oscilloscope #268: OSC failed to launch due to unloaded library on macos

Scottpedia, 2020-09-02:
I was following this [guide](https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope#macos)  to install IIO Oscilloscope on macos 10.15. I got this when I tried to launch osc from the terminal.
```shell
dyld: Library not loaded: @rpath/iio.framework/Versions/0.17/iio
  Referenced from: /Library/Frameworks/ad9361.framework/Versions/0.2/ad9361
  Reason: image not found
Abort trap: 6
``` 

I have installed both iio0.17, iio0.21 and libad9361 with pkg files.

tfcollins, 2020-09-03:
The above error means that the wrong libiio is installed. You must install libiio 0.17 if you are using the pkgs.

-Travis

Scottpedia, 2020-09-03:
I have actually installed the 0.17 several times with the pkg files. And it did not work. I wonder if that’s the problem of the system version the pkgs are built for. I am using macOS 10.15 while these packages for only older versions like 10.14 or 10.13.

tfcollins, 2020-09-03:
I just tried this on my macOS 10.15 and it built fine. Do you get the same error when v0.17 is installed? If this happens can you tell me what is in /Library/Frameworks/iio.framework/Versions ?

-Travis

Scottpedia, 2020-09-03:
@tfcollins I just checked the path and it showed only `0.21` and `Current`. I now remember that I installed 0.21 first. Maybe the system did not install the older version of a package when a new one already exists. Let me try to remove it and install the old one.

Scottpedia, 2020-09-03:
Something funny just happened. I figured out a way to remove 0.21 and install 0.17. And when I tried to launch `osc` from the terminal, it gives out:
```
dyld: Library not loaded: @rpath/iio.framework/Versions/0.21/iio
  Referenced from: /usr/local/bin/osc
  Reason: image not found
Abort trap: 6
```
It seems to me that the `osc` itself requires `0.21` while `libad9361` requires `0.17`. And these two versions cannot coexist.

When I installed `0.21` again, the directory `0.17` under `Versions` disappeared.

tfcollins, 2020-09-03:
osc will require whatever you built it against so you cannot start changing library versions after the fact. Make sure v0.17 is installed then rebuild osc.

-Travis

Scottpedia, 2020-09-03:
I got it. Let me give it a try.

Scottpedia, 2020-09-03:
Thank you, it worked.

closed 2020-09-03 by Scottpedia