analogdevicesinc/iio-oscilloscope #287: Please do not disable peer cert verification during update check

rgov, 2020-11-22:
https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/phone_home.c#L58-L63

This disables certificate validation when checking for updates. Please don't do this. Disabling certificate validation allows a malicious network to modify the update information, and could potentially lead a user to install a malicious update.

This is not necessary because the updates are provided by GitHub which has a valid certificate on its endpoint. If the client is unable to verify GitHub's certificate, it is a misconfiguration of their system. The best behavior in this case is to fail, rather than silently continue.

Also, on macOS Big Sur with libcurl from Homebrew, the API seems to fail if you change this setting, causing the entire app to crash.

dNechita, 2020-11-23:
Thank you for pointing this out!

commodo, 2020-11-23:
closing now, since PR was merged

closed 2020-11-23 by commodo