analogdevicesinc/iio-oscilloscope #292: Installed OSC TDD load fail

tfcollins, 2020-12-07:
After a build and install with cmake (off master), connecting to PlutoSDR causes a failure:

```
tcollins@pop-os  /tmp  osc                                                                   ✔  10416  13:08:09
Failed connecting to remote device: 192.168.86.44
Found plugin: SCPI
Found plugin: FMCOMMS11
Found plugin: FMComms6
Found plugin: FMCADC3
Found plugin: DMM
Found plugin: CN0508
Found plugin: AD9371
Found plugin: AD936X
Found plugin: FMComms1
Found plugin: Motor Control
Found plugin: ad9739a
Found plugin: LIDAR
Found plugin: AD5628-1
Found plugin: AD9371 Advanced
Found plugin: AD6676
Found plugin: Debug
Found plugin: CN0540
Could not find expected iio devices
Found plugin: Spectrum Analyzer
Found plugin: CN0357
Found plugin: DAC Data Manager
Found plugin: AD936X Advanced
Found plugin: CN0511
Failed to load plugin "cf_axi_tdd.so": /usr/local/lib/osc//cf_axi_tdd.so: undefined symbol: iio_widget_save_block_signals_by_data_cb
Found plugin: AD7303
Found plugin: Partial Reconfiguration
Found plugin: FMComms5
Found plugin: DAQ1/2/3
Found plugin: ADRV9009
Loaded plugin: DMM
Updating widgets...
A crash report file has been created at: /home/tcollins/.osc_crash_report
[1]    7800 segmentation fault (core dumped)  osc

```
Running directly from build folder works correctly, so I'm guessing an install target is missing. Removing osc_profile does not help.

OS: Ubuntu 18.04

mhennerich, 2020-12-08:
my guess is that you have build with the makefile before and that cmake installs things into a different location.
Can you try to:
 
$make clean -fd
$git reset --hard
$make install

tfcollins, 2020-12-08:
Yep that did the trick. Thanks Michael. Should we remove the makefile in the future?

-Travis

nunojsa, 2020-12-08:
There's already a [PR](https://github.com/analogdevicesinc/iio-oscilloscope/pull/282) made by @commodo that I would also really like to see merged :smile:

commodo, 2020-12-08:
> There's already a [PR](https://github.com/analogdevicesinc/iio-oscilloscope/pull/282) made by @commodo that I would also really like to see merged smile

once we get to migrate stuff a new CI; i'll take a look there again