analogdevicesinc/iio-oscilloscope #303: Plot update

cuciureansergiu, 2021-01-15:
After changing the sampling frequency for an ADC, the FFT plot is showing the data based on the old sampling rate.
For example,  on a sinewave of 10khz:
default samping rate, reads 10khz
1/2 sampling rate, reads 20khz,
1/4 sampling rate, reads 40khz
Each time the sampling frequency is changed, osc has to be restarted (closing the plot window and opening another one won't fix this).

mhennerich, 2021-01-15:
Where are you seeing this?
I think this is typically handled by the plugins.

cuciureansergiu, 2021-01-15:
Here are some screenshots for a 5 khz sinewave and changing only the sampling frequency for the axi_adc:

[image]
[image]

After i restart osc

[image]

Seems like the graph data is stored once, when osc starts

mhennerich, 2021-01-15:
Typically the plugin calls rx_update_device_sampling_freq(). In your case you use the DEBUG plugin and manipulate attributes.

dNechita, 2021-01-15:
One idea is to have oscplot guess which attribute is for changing the sampling frequency (if plugin doesn't call rx_update_device_sampling_freq() or if there is no plugin). Then oscplot would read that attribute exactly when the start button gets clicked. This way, the sampling rate would get updated next time the capture is started which is better than having to restart the entire aplication.

dbogdan, 2021-01-15:
Can't we check the sampling frequency each time the START button is pressed?

dbogdan, 2021-01-15:
@dNechita , I was thinking the same. :)

mhennerich, 2021-01-15:
I guess we do this already for computing the timeout in capture_setup()

dNechita, 2021-02-23:
A fix for this has been made. Closing this issue.

closed 2021-02-23 by dNechita