analogdevicesinc/iio-oscilloscope #309: iio-oscilloscope is not staring up

76nic, 2021-02-13:
I have installed the iio-oscilloscope on Ubuntu 20.04 following that instructions https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope. All installed just fine. However, the program will not start up. It is asking for the password but after entering the password nothing happens. I also tried to start it up from the command line /usr/local/bin/osc-wrapper, however, that results in the same thing and puts out the following: (osc:8490): Gtk-WARNING **: 10:57:04.025: cannot open display:
Many thanks for your help!
Best regards,
76nic

dNechita, 2021-02-17:
Hi,
Sorry for late response. I will have a look at this today.

dNechita, 2021-02-17:
Meanwhile you should be able to run osc  like this:
sudo /usr/local/bin/osc

76nic, 2021-02-21:
Dear Nechita,
Perfect ! Many thanks for looking into that and helping me out ! It works well...
Best regards,
76nic

closed 2021-02-21 by 76nic

nunojsa, 2024-05-29:
> This issue is still there. It's a problem with LD_LIBRARY_PATH not including /usr/local/lib.

You can change the installation location when running cmake. Place it under /usr/lib and it should work.