analogdevicesinc/iio-oscilloscope #312: Reconsider the install path for org.adi.pkexec.osc.policy

dNechita, 2021-02-22:
After PR: https://github.com/analogdevicesinc/iio-oscilloscope/pull/277, the org.adi.pkexec.osc.policy file is copied (installed) wherever CMAKE_INSTALL_PREFIX says it should, more specifically here: ${CMAKE_INSTALL_PREFIX}/share/polkit-1/actions/
While this is practical as one can control the destination where files are installed, it also creates issues for users that just want to install osc to the default prefix ("/usr/local" for example). With the current implementation the org.adi.pkexec.osc.policy will end-up being installed at: /usr/local/share/polkit-1/actions/ but polkit looks only at: /usr/share/polkit-1/actions/

A user already encountered this issue:
https://github.com/analogdevicesinc/iio-oscilloscope/issues/309

Not sure what's the best solution. One that comes in mind is to create a CMake flag (option) that states where we do an install into the system or we do an install for packaging. And have different strategies for installing org.adi.pkexec.osc.policy.
@mhennerich, @nunojsa, @spacefrogg do you happen to have any thoughts on this?

Thanks!
Dan

spacefrogg, 2021-02-24:
I already fixed this issue (actually, in the same way that you propose) in #281. Nobody bothered to review or merge it, yet.

nunojsa, 2021-02-24:
Thanks for your reply @spacefrogg. But that's not entirely fair as we did reviewed the PR. It just got forgotten int the middle of the CI thing. I will bring the PR up again with some comments...

spacefrogg, 2021-02-24:
I did not mean to be rude. I apologize.

dNechita, 2021-03-04:
@spacefrogg, I apologize about not addressing your PR for such a long time. I will merge your PR today.
I'm closing this issue as your PR answers my question and also solves the issue.
Thank you for contributing!

closed 2021-03-04 by dNechita