analogdevicesinc/iio-oscilloscope #355: Gtk 2.0 and 3.0 conflict when launching

samLeit, 2021-11-09:


nunojsa, 2021-11-09:
Did you already tried to do a clean build? I'm running arch and I cannot reproduce your issue. Things work fine for me. But note that I had to keep gtkdatabox from upgrading as that would break my build.

samLeit, 2021-11-09:
I will try a clean build. With what version of gtkdatabox does your build work? @nunojsa

nunojsa, 2021-11-09:
I have 0.9.3.1 and I'm keeping it from updating to 1.0.0

samLeit, 2021-11-09:
This is the complete output, i am not seeing something that may break it. Also no linking against GTK3, only GTK2.

 Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libgtk-x11-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgdk-x11-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpangocairo-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libatk-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libcairo.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libgdk_pixbuf-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgio-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpangoft2-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpango-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgobject-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libharfbuzz.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libfontconfig.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libfreetype.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libiio.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libosc.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000f (RPATH)              Library rpath: [/home/samuell/workspace/tools/iio-oscilloscope/build:]
 0x000000000000000c (INIT)               0x3000
 0x000000000000000d (FINI)               0x4578
 0x0000000000000019 (INIT_ARRAY)         0x6cc8
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x6cd0
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x3c0
 0x0000000000000005 (STRTAB)             0x1068
 0x0000000000000006 (SYMTAB)             0x498
 0x000000000000000a (STRSZ)              2327 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000003 (PLTGOT)             0x7000
 0x0000000000000002 (PLTRELSZ)           2280 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x1ca0
 0x0000000000000007 (RELA)               0x1ac0
 0x0000000000000008 (RELASZ)             480 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffb (FLAGS_1)            Flags: PIE
 0x000000006ffffffe (VERNEED)            0x1a80
 0x000000006fffffff (VERNEEDNUM)         1
 0x000000006ffffff0 (VERSYM)             0x1980
 0x000000006ffffff9 (RELACOUNT)          3
 0x0000000000000000 (NULL)               0x0

nunojsa, 2021-11-09:
It looks to be fine... This is what ldd gives me:

```
ldd /usr/local/bin/osc
	linux-vdso.so.1 (0x00007ffdc9b51000)
	libgtk-x11-2.0.so.0 => /usr/lib/libgtk-x11-2.0.so.0 (0x00007f4789b3c000)
	libgdk-x11-2.0.so.0 => /usr/lib/libgdk-x11-2.0.so.0 (0x00007f4789a81000)
	libpangocairo-1.0.so.0 => /usr/lib/libpangocairo-1.0.so.0 (0x00007f4789a70000)
	libatk-1.0.so.0 => /usr/lib/libatk-1.0.so.0 (0x00007f4789a47000)
	libcairo.so.2 => /usr/lib/libcairo.so.2 (0x00007f4789928000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/libgdk_pixbuf-2.0.so.0 (0x00007f47898e2000)
	libgio-2.0.so.0 => /usr/lib/libgio-2.0.so.0 (0x00007f4789720000)
	libpangoft2-1.0.so.0 => /usr/lib/libpangoft2-1.0.so.0 (0x00007f4789708000)
	libpango-1.0.so.0 => /usr/lib/libpango-1.0.so.0 (0x00007f47896b6000)
	libgobject-2.0.so.0 => /usr/lib/libgobject-2.0.so.0 (0x00007f478965c000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007f4789526000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007f478944d000)
	libfontconfig.so.1 => /usr/lib/libfontconfig.so.1 (0x00007f47893fc000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f4789332000)
	libiio.so.0 => /usr/lib/libiio.so.0 (0x00007f4789309000)
	libosc.so.0 => /usr/lib/libosc.so.0 (0x00007f47892ac000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f47890e0000)
	libgmodule-2.0.so.0 => /usr/lib/libgmodule-2.0.so.0 (0x00007f47890d9000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f4788f96000)
	libXfixes.so.3 => /usr/lib/libXfixes.so.3 (0x00007f4788f8d000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f4788e49000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f4788e28000)
	libXrender.so.1 => /usr/lib/libXrender.so.1 (0x00007f4788e1b000)
	libXinerama.so.1 => /usr/lib/libXinerama.so.1 (0x00007f4788e16000)
	libXi.so.6 => /usr/lib/libXi.so.6 (0x00007f4788e00000)
	libXrandr.so.2 => /usr/lib/libXrandr.so.2 (0x00007f4788df3000)
	libXcursor.so.1 => /usr/lib/libXcursor.so.1 (0x00007f4788de7000)
	libXcomposite.so.1 => /usr/lib/libXcomposite.so.1 (0x00007f4788de2000)
	libXdamage.so.1 => /usr/lib/libXdamage.so.1 (0x00007f4788ddd000)
	libXext.so.6 => /usr/lib/libXext.so.6 (0x00007f4788dc8000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f4788dac000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007f4788d75000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007f4788d4b000)
	libxcb-render.so.0 => /usr/lib/libxcb-render.so.0 (0x00007f4788d3b000)
	libxcb-shm.so.0 => /usr/lib/libxcb-shm.so.0 (0x00007f4788d36000)
	libpixman-1.so.0 => /usr/lib/libpixman-1.so.0 (0x00007f4788c8e000)
	libjpeg.so.8 => /usr/lib/libjpeg.so.8 (0x00007f4788bfa000)
	libtiff.so.5 => /usr/lib/libtiff.so.5 (0x00007f4788b66000)
	libmount.so.1 => /usr/lib/libmount.so.1 (0x00007f4788b06000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f4788aec000)
	libfribidi.so.0 => /usr/lib/libfribidi.so.0 (0x00007f4788acc000)
	libthai.so.0 => /usr/lib/libthai.so.0 (0x00007f4788ac1000)
	libffi.so.8 => /usr/lib/libffi.so.8 (0x00007f4788ab3000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007f4788a3c000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007f4788a17000)
	libexpat.so.1 => /usr/lib/libexpat.so.1 (0x00007f47889e7000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f47889d4000)
	libbrotlidec.so.1 => /usr/lib/libbrotlidec.so.1 (0x00007f47889c4000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x00007f47889a6000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f478899b000)
	libserialport.so.0 => /usr/lib/libserialport.so.0 (0x00007f4788989000)
	libavahi-client.so.3 => /usr/lib/libavahi-client.so.3 (0x00007f4788975000)
	libavahi-common.so.3 => /usr/lib/libavahi-common.so.3 (0x00007f4788967000)
	libxml2.so.2 => /usr/lib/libxml2.so.2 (0x00007f47887f9000)
	libicudata.so.69 => /usr/lib/libicudata.so.69 (0x00007f4786c9e000)
	libicui18n.so.69 => /usr/lib/libicui18n.so.69 (0x00007f4786977000)
	libicuuc.so.69 => /usr/lib/libicuuc.so.69 (0x00007f4786784000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007f478675b000)
	libgthread-2.0.so.0 => /usr/lib/libgthread-2.0.so.0 (0x00007f4786756000)
	libgtkdatabox-0.9.3.so.1 => /usr/lib/libgtkdatabox-0.9.3.so.1 (0x00007f4786736000)
	libfftw3.so.3 => /usr/lib/libfftw3.so.3 (0x00007f47864fa000)
	libcurl.so.4 => /usr/lib/libcurl.so.4 (0x00007f478645b000)
	libjansson.so.4 => /usr/lib/libjansson.so.4 (0x00007f478644b000)
	libmatio.so.11 => /usr/lib/libmatio.so.11 (0x00007f47863dc000)
	libad9361.so.0 => /usr/lib/libad9361.so.0 (0x00007f478624e000)
[... 9 lines cut ...]
	libdbus-1.so.3 => /usr/lib/libdbus-1.so.3 (0x00007f4786025000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f4785e0f000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f4785df4000)
	libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x00007f4785dc6000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007f4785da4000)
	libssh2.so.1 => /usr/lib/libssh2.so.1 (0x00007f4785d63000)
	libpsl.so.5 => /usr/lib/libpsl.so.5 (0x00007f4785d50000)
	libssl.so.1.1 => /usr/lib/libssl.so.1.1 (0x00007f4785cbe000)
	libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0x00007f47859df000)
	libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2 (0x00007f4785988000)
	libhdf5.so.200 => /usr/lib/libhdf5.so.200 (0x00007f4785597000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007f47854d2000)
	libunistring.so.2 => /usr/lib/libunistring.so.2 (0x00007f4785350000)
	libkrb5.so.3 => /usr/lib/libkrb5.so.3 (0x00007f4785269000)
	libk5crypto.so.3 => /usr/lib/libk5crypto.so.3 (0x00007f4785236000)
	libcom_err.so.2 => /usr/lib/libcom_err.so.2 (0x00007f4785230000)
[... cut ...]
	libsz.so.2 => /usr/lib/libsz.so.2 (0x00007f478520d000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f47851ea000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x00007f47851dd000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f47850a1000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f478507a000)
```
You can also try to run it with strace to see if something odd is going on when loading the libraries... But yeah, most likely there's some issue in your system. A quick workaround that I can suggest is running it from a docker container

k-wozniak, 2021-11-09:
Hi, I am having exactly the same issue. So the problem is not too isolated.

I also got the GtkDatabox 0.9.3.1 but still not working.

`Dynamic section at offset 0x5cd8 contains 43 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libgtk-x11-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgdk-x11-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpangocairo-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libatk-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libcairo.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libgdk_pixbuf-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgio-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpangoft2-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpango-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgobject-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libharfbuzz.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libfontconfig.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libfreetype.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libiio.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libosc.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000f (RPATH)              Library rpath: [/home/kw/Downloads/iio-oscilloscope/build:]
 0x000000000000000c (INIT)               0x3000
 0x000000000000000d (FINI)               0x4578
 0x0000000000000019 (INIT_ARRAY)         0x6cc8
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x6cd0
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x3c0
 0x0000000000000005 (STRTAB)             0x1068
 0x0000000000000006 (SYMTAB)             0x498
 0x000000000000000a (STRSZ)              2316 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000003 (PLTGOT)             0x7000
 0x0000000000000002 (PLTRELSZ)           2280 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x1c90
 0x0000000000000007 (RELA)               0x1ab0
 0x0000000000000008 (RELASZ)             480 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffb (FLAGS_1)            Flags: PIE
 0x000000006ffffffe (VERNEED)            0x1a70
 0x000000006fffffff (VERNEEDNUM)         1
 0x000000006ffffff0 (VERSYM)             0x1974
 0x000000006ffffff9 (RELACOUNT)          3
 0x0000000000000000 (NULL)               0x0
`

k-wozniak, 2021-11-09:
Running ldd I do find two dependencies that might be the root of the problem:
```
libgtk-3.so.0 => /usr/lib/libgtk-3.so.0 (0x00007f4325dc6000)
libgdk-3.so.0 => /usr/lib/libgdk-3.so.0 (0x00007f4325cd2000) 
```
How could I find what is the reason for these dependencies?

nunojsa, 2021-11-09:
yes... that should be your issue. Now you need to figure how are those dependencies getting there :)

samLeit, 2021-11-11:
@k-wozniak 

Found the solution. So the problem was that my installation of gtkdatabox rathter 0.9.* or 1.0 where linking automatically to gtk3.
The solution is to install gtkdatabox  0.9.3.0 from source and do the following steps: (But uninstall gtkdatabox 1.0.0.0 before)
https://ez.analog.com/linux-software-drivers/f/q-a/536533/iio-oscilloscope-build-failed---package-not-found

Its automatically linked to gtk2 then and will cause no problems.
Thanks.

nunojsa, 2021-11-11:
cool... I guess I can then close this?

closed 2021-11-11 by nunojsa