analogdevicesinc/iio-oscilloscope #360: cmake build fails on OSX

rabarar, 2022-01-24:
Has anyone attempted to build this on OSX?

a few thoughts:
- a readme section on pre-reqs for OSX would be helpful
- a list of cmake options on the build would be helpful
- a pre-built binary for those not skilled in using the build system would be helpful

tfcollins, 2022-01-24:
The recommended way is to use homebrew: https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope?rev=1643067915#macos

rabarar, 2022-01-25:
Okay,

that fails too - I have two versions of libiio installed - and the fomula, even if I modify the dependency to specify the tfcollins/formulae/libiio fails

— 

$ brew install --build-from-source ./iio-oscilloscope.rb
Error: Failed to load cask: ./iio-oscilloscope.rb
Cask 'iio-oscilloscope' is unreadable: wrong constant name #<Class:0x00007fc760134b68>
Warning: Treating ./iio-oscilloscope.rb as a formula.
Error: Formulae found in multiple taps: 
       * tfcollins/formulae/libiio
       * pothosware/pothos/libiio

Please use the fully-qualified name (e.g. tfcollins/formulae/libiio) to refer to the formula.

$ brew info tfcollins/formulae/libiio
tfcollins/formulae/libiio: stable 0.21, HEAD
Library for interfacing with local and remote Linux IIO devices
https://analogdevicesinc.github.io/libiio/
/usr/local/Cellar/libiio/0.15 (138 files, 1.9MB) *
  Built from source on 2022-01-23 at 22:22:46
/usr/local/Cellar/libiio/0.21 (25 files, 670.5KB)
  Built from source on 2022-01-23 at 22:21:35
From: https://github.com/tfcollins/homebrew-formulae/blob/HEAD/libiio.rb
License: LGPL-2.1
==> Dependencies
Build: cmake ✔
Required: libserialport ✔, libusb ✔
==> Options
--HEAD
	Install HEAD version

$ brew info pothosware/pothos/libiio
pothosware/pothos/libiio: stable 0.15, HEAD
Library for interfacing with IIO devices.
https://wiki.analog.com/software/linux/docs/iio/iio
/usr/local/Cellar/libiio/0.15 (138 files, 1.9MB) *
  Built from source on 2022-01-23 at 22:22:46
/usr/local/Cellar/libiio/0.21 (25 files, 670.5KB)
  Built from source on 2022-01-23 at 22:21:35
From: https://github.com/pothosware/homebrew-pothos/blob/HEAD/libiio.rb
==> Dependencies
Build: cmake ✔, ninja ✔
==> Options
--HEAD
	Install HEAD version

tfcollins, 2022-01-25:
Remove the pothos variant and retry.

rabarar, 2022-01-25:
removed - however, that too fails as the tap shows the availability of a libiio formula.  you can’t remove every formula on the pothos tap…

tfcollins, 2022-01-25:
Please provide the error.

rabarar, 2022-01-25:
It’s the same error with or without the libiio installed:

$ brew install --build-from-source ./iio-oscilloscope.rb
Error: Failed to load cask: ./iio-oscilloscope.rb
Cask 'iio-oscilloscope' is unreadable: wrong constant name #<Class:0x00007fc760134b68>
Warning: Treating ./iio-oscilloscope.rb as a formula.
Error: Formulae found in multiple taps:
       * tfcollins/formulae/libiio
       * pothosware/pothos/libiio

Please use the fully-qualified name (e.g. tfcollins/formulae/libiio) to refer to the formula.

rabarar, 2022-01-25:
> In fact, this issue persists for other brews - I can’t install gr-iio for the same reason
>
> you need to disambiguate the conflict so one can use the soapysdr pluto driver concurrently with the taps from tfcollins

In the mean time, I removed the tap, installed iio-oscilloscope, and ran it: there are issue with GTK version on OSX:

$ /usr/local/Cellar/iio-oscilloscope/0.11/bin/osc
objc[29523]: Class GdkQuartzView is implemented in both /usr/local/Cellar/gtk+3/3.24.30_1/lib/libgdk-3.0.dylib (0x10f9db128) and /usr/local/Cellar/gtk+/2.24.33/lib/libgdk-quartz-2.0.0.dylib (0x10cc57340). One of the two will be used. Which one is undefined.
objc[29523]: Class GtkQuartzStatusIcon is implemented in both /usr/local/Cellar/gtk+3/3.24.30_1/lib/libgtk-3.0.dylib (0x1106a8db0) and /usr/local/Cellar/gtk+/2.24.33/lib/libgtk-quartz-2.0.0.dylib (0x10d2b5b00). One of the two will be used. Which one is undefined.
objc[29523]: Class GtkClipboardOwner is implemented in both /usr/local/Cellar/gtk+3/3.24.30_1/lib/libgtk-3.0.dylib (0x1106a8d10) and /usr/local/Cellar/gtk+/2.24.33/lib/libgtk-quartz-2.0.0.dylib (0x10d2b5b50). One of the two will be used. Which one is undefined.
objc[29523]: Class GtkDragSourceOwner is implemented in both /usr/local/Cellar/gtk+3/3.24.30_1/lib/libgtk-3.0.dylib (0x1106a8d60) and /usr/local/Cellar/gtk+/2.24.33/lib/libgtk-quartz-2.0.0.dylib (0x10d2b5ba0). One of the two will be used. Which one is undefined.
objc[29523]: Class ResultReceiver is implemented in both /usr/local/Cellar/gtk+3/3.24.30_1/lib/libgtk-3.0.dylib (0x1106a8e28) and /usr/local/Cellar/gtk+/2.24.33/lib/libgtk-quartz-2.0.0.dylib (0x10d2b5c18). One of the two will be used. Which one is undefined.

(osc:29523): Gtk-ERROR **: 12:45:59.482: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

rabarar, 2022-01-25:
also, is there any timeline on supporting gr-iio an gnuradio 3.9?

tfcollins, 2022-01-25:
> you need to disambiguate the conflict so one can use the soapysdr pluto driver concurrently with the taps from tfcollins

Based on how brew works if taps share the same formula name it will break all dependent packages that perform installs. You can only remove a tap to proceed.

tfcollins, 2022-01-25:
> also, is there any timeline on supporting gr-iio an gnuradio 3.9?
> […](#)

Please ask that in a different thread as its unrelated to IIO-Scope development

tfcollins, 2022-01-25:
> In the mean time, I removed the tap, installed iio-oscilloscope, and ran it: there are issue with GTK version on OSX:

This is why there is a special gtk recipe in the IIO-Scope formulae. You cannot use the standard gtk2 databox package.