analogdevicesinc/iio-oscilloscope #365: Sending Captured data by (iio-oscilloscope/plugins/spectrum_analyzer.c ) on AD9361 DAC

waseem-arshad94, 2022-03-05:
I checked iio-oscilloscope/plugins/spectrum_analyzer.c plugin by injecting RF signal on RX1 of the ADRV9361-z7035 . I want to see this peak on AD9361 DAC by enabling RF LOOP BACK by setting ADC path register 0x418 to 0x0 and DAC path register 0x418 to 0x8 .

mhennerich, 2022-03-07:
There are loopback register in the AXI_AD9361 interface HDL core.

But not use the AD9361 loopback modes?
https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#bist_loopback

waseem-arshad94, 2022-03-07:
Thank you for your reply . My problem statement is exactly the same . That it is possible to use loopback as mentioned by you (https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#bist_loopback) while using spectrum analyser . Let me rephrase my statement
1- If enable loopback modes (RF RX---->RF TX) using this link https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#bist_loopback   
2- Then I start /click iio-oscilloscope/plugins/spectrum_analyzer . Will the capture peak /signal be shown on DAC ?

BR

mhennerich, 2022-03-07:
This plugin only sweeps the RX LO.
But yes you will see whatever is on the RX loopbacked to the TX.

waseem-arshad94, 2022-03-07:
Thanks for your valuable response! Actually I want to sweep the RX LO from 70 MHz to 6 GHz in loopback, and want to see the result on spectrum analyzer. I will put Signal generator in sweep mode from 70 MHz to 6 GHz and want to see the result on Spectrum Analyzer. I go through these (https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#bist_loopback) but not reach the end result. Can you give me little bit explanation how I can done this job in loopback mode?
thanks

waseem-arshad94, 2022-03-07:
[image]

[image]

Now I injected RF 507 MHz that is captured by spectrum analyzer iio oscilloscope but at there is no output on the external spectrum analyzer connected to TX1 port . 
noted: RF RX- RF TX loopback is enabled

mhennerich, 2022-03-07:
What is the TX LO frequency?

waseem-arshad94, 2022-03-07:
TX LO is 2400MHz by default
I understand that loop back will work for 1x RX LO & TX LO ( with RF bandwidth 56 MHz ) but is it possible that i do loopback for whole range 70MHz-6GHz

waseem-arshad94, 2022-03-07:
The peak is captured when I clicked start sweep button on iio oscilloscope

mhennerich, 2022-03-07:
> I understand that loop back will work for 1x RX LO & TX LO ( with RF bandwidth 56 MHz ) but is it possible that i do loopback for whole range 70MHz-6GHz

No - The device has max RF bandwidth of 56 MHz

waseem-arshad94, 2022-03-08:
But in loopback it works for iio-oscilloscope for entire range 70 MHz to 6 GHz. But its not displaying on spectrum analyzer how its different from iio-oscilloscope. Any setting I am missing?

mhennerich, 2022-03-08:
> But in loopback it works for iio-oscilloscope for entire range 70 MHz to 6 GHz. But its not displaying on spectrum analyzer how its different from iio-oscilloscope. Any setting I am missing?

It's not a single capture. It captures some bandwidth, computes the FFT, updates the LO and starts to capture the next buffer. The FFTs are then concatenated.

tfcollins, 2022-03-08:
More details: https://wiki.analog.com/resources/tools-software/linux-software/spectrum_analyzer_plugin

waseem-arshad94, 2022-03-22:
Thank you Sir @tfcollins and Sir @mhennerich . My objective is that combine spectrum analyzer fuction ( peak search , peak detector ) and DDS/DAC . I think I have alot of work to do but this will help me alot

closed 2023-12-14 by dNechita