analogdevicesinc/iio-oscilloscope #370: Ubuntu 22.04 cmake error

octaplexsys, 2022-04-22:
Can't build from sources under latest version of linux kernel (Ubuntu 22.04 LTS)
Step to reproduce : 
According to documentation on page 
https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope#installation
for step 5 on cmake execution activity ,get some error

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

mhennerich, 2022-04-25:
Can you post the entire error message?
Maybe it's due to GTK2?
We're in progress updating to GTK3...

huguitovi, 2022-07-02:
the first problem I encountered was that it is not indicated in the tutorial on the page https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope that the following packages should be installed in Debian

> libad9361-dev 
> libiio-dev

with the comanod:
```
sudo apt install libad9361-dev libiio-dev
```

this improved the error output but I still can't find another missing package. 
the output of the cmake below 

```
~/iio-oscilloscope/build$ cmake ../ && make -j $(nproc)
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'glib-2.0'
--   Found glib-2.0, version 2.58.3
-- Checking for module 'gtk+-2.0'
--   Found gtk+-2.0, version 2.24.32
-- Checking for module 'gthread-2.0'
--   Found gthread-2.0, version 2.58.3
-- Checking for module 'gtkdatabox'
--   Found gtkdatabox, version 0.9.3.1
-- Checking for module 'fftw3'
--   Found fftw3, version 3.3.8
-- Checking for module 'libxml-2.0'
--   Found libxml-2.0, version 2.9.4
-- Checking for module 'libcurl'
--   Found libcurl, version 7.64.0
-- Checking for module 'jansson'
--   Found jansson, version 2.12
-- Checking for module 'matio'
--   Found matio, version 1.5.13
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/focus/iio-oscilloscope/LIBAD9166_INCLUDE_DIRS
   used as include directory in directory /home/focus/iio-oscilloscope
/home/focus/iio-oscilloscope/plugins/LIBAD9166_INCLUDE_DIRS
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
   used as include directory in directory /home/focus/iio-oscilloscope/plugins
[... 46 lines cut ...]
```

details  of PC

```
       _,met$$$$$gg.          jhonddoe@anon 
    ,g$$$$$$$$$$$$$$$P.       ----------- 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 10 (buster) x86_64 
 ,$$P'              `$$$.     Host: 530U3C/530U4C 0.1 
',$$P       ,ggs.     `$$b:   Kernel: 4.19.0-5-amd64 
`d$$'     ,$P"'   .    $$$    Uptime: 1 hour, 9 mins 
 $$P      d$'     ,    $$P    Packages: 2152 (dpkg) 
 $$:      $$.   -    ,d$$'    Shell: bash 5.0.3 
 $$;      Y$b._   _,d$P'      Resolution: 1366x768 
 Y$$.    `.`"Y$$$$P"'         DE: Xfce 
 `$$b      "-.__              WM: Xfwm4 
  `Y$$                        WM Theme: Default 
   `Y$$.                      Theme: Xfce-saltlake [GTK2], Adwaita [GTK3] 
     `$$b.                    Icons: Tango [GTK2], Adwaita [GTK3] 
       `Y$$b.                 Terminal: xfce4-terminal 
          `"Y$b._             Terminal Font: Monospace 12 
              `"""            CPU: Intel i5-3317U (4) @ 2.600GHz 
                              GPU: Intel 3rd Gen Core processor Graphics Contro 
                              Memory: 1256MiB / 5743MiB 
```

huguitovi, 2022-07-02:
after finding the missing package by running the script from the page indicated in the previous post for the [ADI upgrade][https://raw.githubusercontent.com/analogdevicesinc/linux_image_ADI-scripts/master/adi_update_tools.sh] and installing the [libiio dependencies][https://wiki.analog.com/resources/tools-software/linux-software/libiio#how_to_build_it] as well, with the command:
```
sudo apt-get install libxml2 libxml2-dev bison flex libcdk5-dev cmake
sudo apt-get install libaio-dev libusb-1.0-0-dev libserialport-dev libxml2-dev libavahi-client-dev doxygen graphviz
```

it seems that I found the missing package, however it still does not compile, the compile message was as follows:

```
Scanning dependencies of target osc
[  1%] Building C object CMakeFiles/osc.dir/osc.c.o
[  2%] Building C object CMakeFiles/osc.dir/oscplot.c.o
[  3%] Building C object CMakeFiles/osc.dir/datatypes.c.o
[  4%] Building C object CMakeFiles/osc.dir/iio_widget.c.o
[  6%] Building C object CMakeFiles/osc.dir/iio_utils.c.o
[  7%] Building C object CMakeFiles/osc.dir/fru.c.o
/home/focus/iio-oscilloscope/iio_utils.c: In function ‘get_iio_device_label_or_name’:
/home/focus/iio-oscilloscope/iio_utils.c:148:7: error: implicit declaration of function ‘iio_device_get_label’; did you mean ‘iio_device_get_name’? [-Werror=implicit-function-declaration]
  id = iio_device_get_label(dev);
       ^~~~~~~~~~~~~~~~~~~~
       iio_device_get_name
/home/focus/iio-oscilloscope/iio_utils.c:148:5: error: assignment to ‘const char *’ from ‘int’ makes pointer from integer without a cast [-Werror=int-conversion]
  id = iio_device_get_label(dev);
     ^
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/osc.dir/build.make:115: CMakeFiles/osc.dir/iio_utils.c.o] Error 1
make[2]: *** Se espera a que terminen otras tareas....
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/osc.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

tfcollins, 2022-07-03:
The packages in the distribution repo are too old. As noted in the instructions on the wiki you need to build and install libiio and libad9361 from source.

huguitovi, 2022-07-03:
that's correct, as it says in the tutorial, I installed `libiio` and `libini` from source and everything worked, then I had to add a step when installing the shortcut that installs when you run the "`make install`" directs to a different application than `/usr/local/osc `so I corrected that and it ended up working, again thank you very much Travis.

big hug
VH

AlexMclellan, 2022-11-23:
To help anyone else out, I also had an issue building on Ubuntu 22.04, after following the guide. Attempting to build master, I would get these errors;

> 
> [  2%] Building C object CMakeFiles/osc.dir/oscplot.c.o
> In file included from /usr/include/gtkdatabox.h:24,
>                  from /home/alex/repos/iio-oscilloscope-master/oscplot.c:9:
> /usr/include/gtkdatabox_graph.h:100:38: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   100 |                                      GdkRGBA * color);
>       |                                      ^~~~~~~
>       |                                      GdkGC
> In file included from /usr/include/gtkdatabox.h:24,
>                  from /home/alex/repos/iio-oscilloscope-master/osc.c:8:
> /usr/include/gtkdatabox_graph.h:100:38: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   100 |                                      GdkRGBA * color);
>       |                                      ^~~~~~~
>       |                                      GdkGC
> /usr/include/gtkdatabox_graph.h:101:4: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   101 |    GdkRGBA *gtk_databox_graph_get_color (GtkDataboxGraph * graph);
>       |    ^~~~~~~
>       |    GdkGC
> /usr/include/gtkdatabox_graph.h:101:4: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   101 |    GdkRGBA *gtk_databox_graph_get_color (GtkDataboxGraph * graph);
>       |    ^~~~~~~
>       |    GdkGC
> /usr/include/gtkdatabox_graph.h:104:38: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   104 |                                      GdkRGBA * rgba);
>       |                                      ^~~~~~~
>       |                                      GdkGC
> /usr/include/gtkdatabox_graph.h:104:38: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   104 |                                      GdkRGBA * rgba);
>       |                                      ^~~~~~~
>       |                                      GdkGC
> /usr/include/gtkdatabox_graph.h:105:4: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   105 |    GdkRGBA *gtk_databox_graph_get_rgba (GtkDataboxGraph * graph);
>       |    ^~~~~~~
>       |    GdkGC
> /usr/include/gtkdatabox_graph.h:105:4: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   105 |    GdkRGBA *gtk_databox_graph_get_rgba (GtkDataboxGraph * graph);
>       |    ^~~~~~~
>       |    GdkGC
> In file included from /home/alex/repos/iio-oscilloscope-master/oscplot.c:10:
> /usr/include/gtkdatabox_grid.h:99:43: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>    99 |                                           GdkRGBA * color, guint size);
>       |                                           ^~~~~~~
>       |                                           GdkGC
> In file included from /home/alex/repos/iio-oscilloscope-master/osc.c:9:
> /usr/include/gtkdatabox_grid.h:99:43: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>    99 |                                           GdkRGBA * color, guint size);
>       |                                           ^~~~~~~
>       |                                           GdkGC
> /usr/include/gtkdatabox_grid.h:101:43: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   101 |                                           GdkRGBA * color, guint size);
>       |                                           ^~~~~~~
>       |                                           GdkGC
> /usr/include/gtkdatabox_grid.h:101:43: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>   101 |                                           GdkRGBA * color, guint size);
>       |                                           ^~~~~~~
>       |                                           GdkGC
> In file included from /home/alex/repos/iio-oscilloscope-master/oscplot.c:11:
> /usr/include/gtkdatabox_points.h:83:45: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>    83 |                                             GdkRGBA * color, gint size);
>       |                                             ^~~~~~~
>       |                                             GdkGC
> In file included from /home/alex/repos/iio-oscilloscope-master/osc.c:10:
> /usr/include/gtkdatabox_points.h:83:45: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>    83 |                                             GdkRGBA * color, gint size);
>       |                                             ^~~~~~~
>       |                                             GdkGC
> /usr/include/gtkdatabox_points.h:88:25: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
>    88 |                         GdkRGBA * color, gint size);
[... 65 lines cut ...]
>  3428 |                         graph = gtk_databox_lines_new(transform->y_axis_size,
>       |                               ^
> /home/alex/repos/iio-oscilloscope-master/oscplot.c: In function ‘add_grid’:
> /home/alex/repos/iio-oscilloscope-master/oscplot.c:4179:30: error: implicit declaration of function ‘gtk_databox_grid_array_new’; did you mean ‘gtk_databox_grid_get_vlines’? [-Werror=implicit-function-declaration]
>  4179 |                 priv->grid = gtk_databox_grid_array_new (25, 14, priv->gridy, priv->gridx, &color_grid, 1);
>       |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
>       |                              gtk_databox_grid_get_vlines
[... cut ...]
>  4194 |                 priv->grid = gtk_databox_grid_array_new (18, 5, priv->gridy, priv->gridx, &color_grid, 1);
>       |                            ^
> cc1: all warnings being treated as errors
> make[2]: *** [CMakeFiles/osc.dir/build.make:90: CMakeFiles/osc.dir/oscplot.c.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/osc.dir/all] Error 2
> make: *** [Makefile:136: all] Error 2

It appears that I can build the `gtk3transition-master` branch, but running it generates errors like `g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed` for me.

dNechita, 2022-11-23:
It's that an error build? Or just a warning?

AlexMclellan, 2022-11-23:
It was a warning when running a build from the `gtk3transition-master` branch, but I've managed to get it to run (user error). However I wasn't able to get anything to build from master, I guess if `gtk3transition-master` is mostly there, I can just use that.

mhennerich, 2023-03-08:
gtk3transition-master was updated and merged, this can be closed now.

closed 2023-03-08 by mhennerich