analogdevicesinc/iio-oscilloscope #372: iio-oscilloscope at Windows10 not launching

younghosuh, 2022-05-05:
I downloaded IIO-Oscilloscope - v0.14-master the first time and installed it into the Windows10 system. I double-clicked osc.exe or the desktop shortcut but nothing happened. Can you please advise what went wrong?

Thank you.

dNechita, 2022-07-20:
Does it work if you download an official release? That should have a known publisher.

younghosuh, 2022-07-20:
I solved the issue. The problem was USB connection. After I switched other USB hub for the connection, everything worked out. 
Thanks,

Sorry for the brevity.
Youngho

tuancoiz4, 2022-12-25:
I am having the same problem when open the osc.exe file. I have tried to change the USB port of the ADALM-Pluto but still cannot open the app

OK2AMA, 2023-01-20:
Hi, I have same problem too. Win 10, in past it works well, but now even screen is not shown. Reinstall and delete osc_profile did not help.