analogdevicesinc/iio-oscilloscope #384: Compilation Fails with Newest libgtkdatabox-dev

chancecardona, 2022-07-15:
On Debian bookworm amd64, with Testing PPA's enabled, the iio-oscilloscope does not compile correctly with the newest version of software. The error is due to the gtkdatabox versioning, despite passing the checks in the adi_update_tools.sh.
**Expected Behavior:**
Running CMake:
```
-- Checking for module 'gtkdatabox'
--   Found gtkdatabox, version 0.9.3.1
-- Checking for module 'fftw3'
--   Found fftw3, version 3.3.8
-- Checking for module 'libxml-2.0'
--   Found libxml-2.0, version 2.9.14
-- Checking for module 'libcurl'
--   Found libcurl, version 7.83.1
-- Checking for module 'jansson'
--   Found jansson, version 2.14
-- Checking for module 'matio'
--   Found matio, version 1.5.23
-- Configuring done
-- Generating done
-- Build files have been written to: /home/chance/iio-oscilloscope/build

```

Make:
```
＄ make -j $(nproc)
[  4%] Building C object CMakeFiles/osc.dir/datatypes.c.o
[  4%] Building C object CMakeFiles/osc.dir/oscplot.c.o
[  4%] Building C object CMakeFiles/osc.dir/osc.c.o
[  4%] Building C object CMakeFiles/osc.dir/iio_widget.c.o
[  6%] Building C object CMakeFiles/osc.dir/iio_utils.c.o
[  7%] Building C object CMakeFiles/osc.dir/fru.c.o
[  8%] Building C object CMakeFiles/osc.dir/dialogs.c.o
[  9%] Building C object CMakeFiles/osc.dir/trigger_dialog.c.o
[ 10%] Building C object CMakeFiles/osc.dir/xml_utils.c.o
[ 12%] Building C object CMakeFiles/osc.dir/libini/libini.c.o
[ 13%] Building C object CMakeFiles/osc.dir/libini2.c.o
[ 14%] Building C object CMakeFiles/osc.dir/phone_home.c.o
[ 15%] Building C object CMakeFiles/osc.dir/plugins/dac_data_manager.c.o
[ 16%] Building C object CMakeFiles/osc.dir/plugins/fir_filter.c.o
[ 18%] Building C object CMakeFiles/osc.dir/eeprom.c.o
[ 19%] Building C object CMakeFiles/osc.dir/osc_preferences.c.o
[ 20%] Linking C shared library libosc.so
[ 20%] Built target osc
[ 22%] Building C object plugins/CMakeFiles/fmcomms1.dir/fmcomms1.c.o
[ 22%] Building C object plugins/CMakeFiles/generic_dac.dir/generic_dac.c.o
[ 24%] Building C object plugins/CMakeFiles/fmcomms2.dir/fmcomms2.c.o
[ 25%] Building C object CMakeFiles/oscmain.dir/oscmain.c.o
[ 26%] Linking C shared library generic_dac.so
[ 27%] Linking C executable osc
[ 27%] Built target generic_dac
[ 28%] Building C object plugins/CMakeFiles/fmcomms2_adv.dir/fmcomms2_adv.c.o
[ 28%] Built target oscmain
[ 30%] Building C object plugins/CMakeFiles/fmcomms5.dir/fmcomms5.c.o
[ 31%] Linking C shared library fmcomms2_adv.so
[ 32%] Linking C shared library fmcomms2.so
[ 33%] Linking C shared library fmcomms1.so
[ 33%] Built target fmcomms2
[ 33%] Built target fmcomms2_adv
[ 33%] Built target fmcomms1
[ 36%] Building C object plugins/CMakeFiles/fmcomms6.dir/fmcomms6.c.o
[ 36%] Building C object plugins/CMakeFiles/fmcomms11.dir/fmcomms11.c.o
[ 37%] Building C object plugins/CMakeFiles/ad9371.dir/ad9371.c.o
[ 38%] Linking C shared library fmcomms6.so
[ 39%] Linking C shared library fmcomms11.so
[ 40%] Linking C shared library fmcomms5.so
[ 40%] Built target fmcomms6
[ 40%] Built target fmcomms11
[ 42%] Building C object plugins/CMakeFiles/ad9371_adv.dir/ad9371_adv.c.o
[ 42%] Built target fmcomms5
[ 43%] Building C object plugins/CMakeFiles/adrv9009.dir/adrv9009.c.o
[... 144 lines cut ...]
/usr/include/gtkdatabox_markers.h:164:25: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
  164 |                         GdkRGBA * color, guint size,
      |                         ^~~~~~~
      |                         GdkGC
/usr/include/gtkdatabox_markers.h:164:25: error: unknown type name ‘GdkRGBA’; did you mean ‘GdkGC’?
  164 |                         GdkRGBA * color, guint size,
      |                         ^~~~~~~
      |                         GdkGC
make[2]: *** [CMakeFiles/osc.dir/build.make:76: CMakeFiles/osc.dir/osc.c.o] Error 1
/home/chance/iio-oscilloscope/oscplot.c: In function ‘markers_init’:
/home/chance/iio-oscilloscope/oscplot.c:2651:36: error: implicit declaration of function ‘gtk_databox_markers_new’; did you mean ‘gtk_databox_ruler_new’? [-Werror=implicit-function-declaration]
 2651 |                 markers[i].graph = gtk_databox_markers_new(1, &markers[i].x, &markers[i].y, &color_marker,
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~
      |                                    gtk_databox_ruler_new
/home/chance/iio-oscilloscope/oscplot.c:2651:34: error: assignment to ‘GtkDataboxGraph *’ from ‘int’ makes pointer from integer without a cast [-Werror=int-conversion]
 2651 |                 markers[i].graph = gtk_databox_markers_new(1, &markers[i].x, &markers[i].y, &color_marker,
      |                                  ^

...
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/osc.dir/build.make:90: CMakeFiles/osc.dir/oscplot.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/osc.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```
The resolution was found by installing the stable versions of the package, namely libgtkdatabox0_0.9.3.1-2 and libgtkdatabox-dev_0.9.3.1-2

dNechita, 2022-07-20:
Hi,
We are currently in the process of migrating iio-oscilloscope from GTK2 to GTK3. This also means it will move from libgtkdatabox 0.9 to 1.0.
Once the migration is finished you won't be encountering this kind of error.

ly2ss, 2022-11-04:
Is there any way to run osc on Ubuntu 22.04LTS while migration lasts?  It has  libgtkdatabox 1.0. 
f.e. install libgtkdatabox 0.9 and link against it.

nunojsa, 2022-11-04:
> Is there any way to run osc on Ubuntu 22.04LTS while migration lasts? It has libgtkdatabox 1.0.
f.e. install libgtkdatabox 0.9 and link against it.

You can get the [source code](https://sourceforge.net/p/gtkdatabox/git/ci/0.9.1.3/tree/) and try to compile it yourself. But do you have any hard requirement to run iut un 22.04LTS? I would just recommend to use docker and create an image based on ubuntu 20.04 already with osc built and run it from the container... It is possible to run GUI apps inside a docker container

dNechita, 2022-11-04:
Just a note. We are close to finishing the transition to GTK3. It should be completed by the middle of this month.

sunwardflyer, 2023-01-15:
@dNechita I'm new to the world of Linux apps and just spent 2.5 days trying to install the IIO osc application. I was having the same issue with the gtkdatabox that the OP of this issue reported. All my issues were resolved by downgrading my Ubuntu install from 22.04 to 20.04. I was hoping could you explain to a noob why I was able to build on the older Ubuntu, but not the newer one?

dNechita, 2023-01-27:
> 

On Ubuntu 20.04, when you install gtkdatabox via apt, you will receive a package with version 0.9.whatever. This package is based on GTK2. Since gtkdatabox is a widget used within iio-osc, osc also needs to be based on GTK2, and it is. So build succeeds.
On Ubuntu 22.04 you are no longer getting a gtkdatabox package based on GTK2 but on GTK3. Version will be 1.whatever. 
So in this case the build process of iio-osc will notice gtkdatabox has GTK3 dependencies and will try to build iio-osc with GTK3 as well. But iio-osc code was written for GTK2 so it won't compile.

sunwardflyer, 2023-01-28:
Thank you @dNechita !

John132-LI, 2023-04-11:
Hello, I encounter a similar issue.I have follow https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope to install ,but when "sudo make install", the error occurs.Here's the log

In file included from /usr/include/gtkdatabox.h:24,
                 from /home/user/Desktop/iio-oscilloscope/osc.c:8:
/usr/include/gtkdatabox_graph.h:86:7: error: unknown type name ‘GdkGC’
   86 |       GdkGC* (*create_gc) (GtkDataboxGraph * graph, GtkDatabox * box);
      |       ^~~~~
/usr/include/gtkdatabox_graph.h:91:4: error: unknown type name ‘GdkGC’; did you mean ‘GdkRGBA’?
   91 |    GdkGC* gtk_databox_graph_get_gc (GtkDataboxGraph * graph);
      |    ^~~~~
      |    GdkRGBA
/usr/include/gtkdatabox_graph.h:92:60: error: unknown type name ‘GdkGC’; did you mean ‘GdkRGBA’?
   92 |    void gtk_databox_graph_set_gc (GtkDataboxGraph * graph, GdkGC *gc);
      |                                                            ^~~~~
      |                                                            GdkRGBA
/usr/include/gtkdatabox_graph.h:111:4: error: unknown type name ‘GdkGC’; did you mean ‘GdkRGBA’?
  111 |    GdkGC* gtk_databox_graph_create_gc (GtkDataboxGraph * graph, GtkDatabox * box);
      |    ^~~~~
      |    GdkRGBA
In file included from /home/user/Desktop/iio-oscilloscope/osc.c:8:
/usr/include/gtkdatabox.h:196:1: error: unknown type name ‘GdkPixmap’; did you mean ‘GdkPixbuf’?
  196 | GdkPixmap* gtk_databox_get_backing_pixmap(GtkDatabox * box);
      | ^~~~~~~~~
      | GdkPixbuf
make[2]: *** [CMakeFiles/osc.dir/build.make:63: CMakeFiles/osc.dir/osc.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:183: CMakeFiles/osc.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

Could someone tell me how to deal with it?Much appreciation.