analogdevicesinc/iio-oscilloscope #392: The 0.15 release is broken: error: argument unused during compilation: '-rdynamic'

yurivict, 2022-12-09:
```
[  7% 7/82] /usr/local/libexec/ccache/cc -DFRU_FILES=\"/lib/fmc-tools/\" -DGTK_DISABLE_DEPRECATED -DOSC_VERSION=\"v0.8\" -D_GNU_SOURCE -D_POSIX_C_SOURCE=200809L -Dlinux -Dosc_EXPORTS -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/gtk-2.0 -I/usr/local/include/pango-1.0 -I/usr/local/include/harfbuzz -I/usr/local/include/freetype2 -I/usr/local/include/libpng16 -I/usr/local/include/fribidi -I/usr/local/include/cairo -I/usr/local/include/pixman-1 -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/atk-1.0 -I/usr/local/include/gtkdatabox -I/usr/local/include/libxml2 -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -DNDEBUG -fPIC -Wall -Wextra -Werror -Wno-unused-parameter -Wno-error=unused-but-set-variable -Wno-error=unused-function -Wimplicit-fallthrough -Wvla -std=gnu90 -g -O2 -rdynamic -MD -MT CMakeFiles/osc.dir/osc.c.o -MF CMakeFiles/osc.dir/osc.c.o.d -o CMakeFiles/osc.dir/osc.c.o -c /disk-samsung/freebsd-ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.15-master/osc.c
FAILED: CMakeFiles/osc.dir/osc.c.o 
/usr/local/libexec/ccache/cc -DFRU_FILES=\"/lib/fmc-tools/\" -DGTK_DISABLE_DEPRECATED -DOSC_VERSION=\"v0.8\" -D_GNU_SOURCE -D_POSIX_C_SOURCE=200809L -Dlinux -Dosc_EXPORTS -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/gtk-2.0 -I/usr/local/include/pango-1.0 -I/usr/local/include/harfbuzz -I/usr/local/include/freetype2 -I/usr/local/include/libpng16 -I/usr/local/include/fribidi -I/usr/local/include/cairo -I/usr/local/include/pixman-1 -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/atk-1.0 -I/usr/local/include/gtkdatabox -I/usr/local/include/libxml2 -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -DNDEBUG -fPIC -Wall -Wextra -Werror -Wno-unused-parameter -Wno-error=unused-but-set-variable -Wno-error=unused-function -Wimplicit-fallthrough -Wvla -std=gnu90 -g -O2 -rdynamic -MD -MT CMakeFiles/osc.dir/osc.c.o -MF CMakeFiles/osc.dir/osc.c.o.d -o CMakeFiles/osc.dir/osc.c.o -c /disk-samsung/freebsd-ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.15-master/osc.c
cc: error: argument unused during compilation: '-rdynamic' [-Werror,-Wunused-command-line-argument]
[  8% 7/82] /usr/local/libexec/ccache/cc -DFRU_FILES=\"/lib/fmc-tools/\" -DGTK_DISABLE_DEPRECATED -DOSC_VERSION=\"v0.8\" -D_GNU_SOURCE -D_POSIX_C_SOURCE=200809L -Dlinux -Dosc_EXPORTS -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/gtk-2.0 -I/usr/local/include/pango-1.0 -I/usr/local/include/harfbuzz -I/usr/local/include/freetype2 -I/usr/local/include/libpng16 -I/usr/local/include/fribidi -I/usr/local/include/cairo -I/usr/local/include/pixman-1 -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/atk-1.0 -I/usr/local/include/gtkdatabox -I/usr/local/include/libxml2 -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -DNDEBUG -fPIC -Wall -Wextra -Werror -Wno-unused-parameter -Wno-error=unused-but-set-variable -Wno-error=unused-function -Wimplicit-fallthrough -Wvla -std=gnu90 -g -O2 -rdynamic -MD -MT CMakeFiles/osc.dir/oscplot.c.o -MF CMakeFiles/osc.dir/oscplot.c.o.d -o CMakeFiles/osc.dir/oscplot.c.o -c /disk-samsung/freebsd-ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.15-master/oscplot.c
FAILED: CMakeFiles/osc.dir/oscplot.c.o 
/usr/local/libexec/ccache/cc -DFRU_FILES=\"/lib/fmc-tools/\" -DGTK_DISABLE_DEPRECATED -DOSC_VERSION=\"v0.8\" -D_GNU_SOURCE -D_POSIX_C_SOURCE=200809L -Dlinux -Dosc_EXPORTS -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/gtk-2.0 -I/usr/local/include/pango-1.0 -I/usr/local/include/harfbuzz -I/usr/local/include/freetype2 -I/usr/local/include/libpng16 -I/usr/local/include/fribidi -I/usr/local/include/cairo -I/usr/local/include/pixman-1 -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/atk-1.0 -I/usr/local/include/gtkdatabox -I/usr/local/include/libxml2 -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -DNDEBUG -fPIC -Wall -Wextra -Werror -Wno-unused-parameter -Wno-error=unused-but-set-variable -Wno-error=unused-function -Wimplicit-fallthrough -Wvla -std=gnu90 -g -O2 -rdynamic -MD -MT CMakeFiles/osc.dir/oscplot.c.o -MF CMakeFiles/osc.dir/oscplot.c.o.d -o CMakeFiles/osc.dir/oscplot.c.o -c /disk-samsung/freebsd-ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.15-master/oscplot.c
cc: error: argument unused during compilation: '-rdynamic' [-Werror,-Wunused-command-line-argument]
```

```-rdynamic``` is a link flag but it is supplied during the compilation phase which breaks the compiler.

clang-14
FreeBSD 13.1

nunojsa, 2022-12-12:
I guess we can use `-Wl,--export-dynamic` to fix it...

dNechita, 2022-12-13:
Hi!
If you have the time, can you try this out?
https://github.com/analogdevicesinc/iio-oscilloscope/pull/394

closed 2023-12-14 by dNechita