analogdevicesinc/iio-oscilloscope #404: Create .deb packages

dNechita, 2023-02-07:
It would be nice if we could generate .deb packages for IIO-Oscilloscope.
It could be done once for each release (manually or automated). Additionally, the package could be generated for each CI build job.

gastmaier, 2023-03-21:
I created a [pull request to generate flatpaks](https://github.com/analogdevicesinc/iio-oscilloscope/pull/421).
About the debian package, iio-oscilloscope is dynamically linked to these libraries:

<details>
<summary>
shared library dependencies 
</summary>

```bash
[📦 org.adi.IIO-Oscilloscope lib]$ ldd libosc.so
	linux-vdso.so.1 (0x00007fff9596b000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f5744a7a000)
	libgtk-3.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-3.so.0 (0x00007f5744200000)
	libgdk-3.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-3.so.0 (0x00007f57440f2000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f5744a4b000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f57449e8000)
	libgtkdatabox.so.1 => /app/lib/libgtkdatabox.so.1 (0x00007f57440cc000)
	libfftw3.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3.so.3 (0x00007f5743ed0000)
	libiio.so.0 => /app/lib/libiio.so.0 (0x00007f5743ead000)
	libxml2.so.2 => /app/lib/libxml2.so.2 (0x00007f5743d3d000)
	libcurl.so.4 => /usr/lib/x86_64-linux-gnu/libcurl.so.4 (0x00007f5743c9a000)
	libjansson.so.4 => /app/lib/libjansson.so.4 (0x00007f5743c89000)
	libmatio.so.11 => /app/lib/libmatio.so.11 (0x00007f5743c18000)
	libc.so.6 => /usr/lib/x86_64-linux-gnu/libc.so.6 (0x00007f5743a00000)
	libpcre2-8.so.0 => /usr/lib/x86_64-linux-gnu/libpcre2-8.so.0 (0x00007f5743965000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f57449dd000)
	libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007f574376a000)
	libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f5743758000)
	libpango-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007f57436ee000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f57435e5000)
	libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f57434bf000)
	libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007f57434a3000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f5743455000)
	libfribidi.so.0 => /usr/lib/x86_64-linux-gnu/libfribidi.so.0 (0x00007f5743435000)
	libcairo-gobject.so.2 => /usr/lib/x86_64-linux-gnu/libcairo-gobject.so.2 (0x00007f5743429000)
	libatk-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007f57433ff000)
	libepoxy.so.0 => /usr/lib/x86_64-linux-gnu/libepoxy.so.0 (0x00007f57432d7000)
	libm.so.6 => /usr/lib/x86_64-linux-gnu/libm.so.6 (0x00007f57431f4000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f57431e0000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f5743098000)
	libatk-bridge-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-bridge-2.0.so.0 (0x00007f574305a000)
	libtracker-sparql-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libtracker-sparql-3.0.so.0 (0x00007f5742f8b000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f5742f83000)
	libxkbcommon.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f5742f3c000)
	libwayland-client.so.0 => /usr/lib/x86_64-linux-gnu/libwayland-client.so.0 (0x00007f5742f28000)
	libwayland-cursor.so.0 => /usr/lib/x86_64-linux-gnu/libwayland-cursor.so.0 (0x00007f5742f1d000)
	libwayland-egl.so.1 => /usr/lib/x86_64-linux-gnu/libwayland-egl.so.1 (0x00007f5743c0d000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f5742f09000)
	libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f5742efc000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f5742ef7000)
	libXcomposite.so.1 => /usr/lib/x86_64-linux-gnu/libXcomposite.so.1 (0x00007f5742ef2000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f5742ee5000)
	libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007f5742ede000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f5742ea4000)
	libjpeg.so.62 => /usr/lib/x86_64-linux-gnu/libjpeg.so.62 (0x00007f5742dfc000)
	libffi.so.8 => /usr/lib/x86_64-linux-gnu/libffi.so.8 (0x00007f5742dec000)
	libz.so.1 => /usr/lib/x86_64-linux-gnu/libz.so.1 (0x00007f5742dd1000)
	libusb-1.0.so.0 => /app/lib/libusb-1.0.so.0 (0x00007f5742daf000)
	libavahi-client.so.3 => /app/lib/libavahi-client.so.3 (0x00007f5742d9b000)
	libavahi-common.so.3 => /app/lib/libavahi-common.so.3 (0x00007f5742d8c000)
	liblzma.so.5 => /usr/lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f5742d62000)
	libnghttp2.so.14 => /usr/lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007f5742d31000)
	libssl.so.3 => /usr/lib/x86_64-linux-gnu/libssl.so.3 (0x00007f5742c88000)
	libcrypto.so.3 => /usr/lib/x86_64-linux-gnu/libcrypto.so.3 (0x00007f5742800000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5744c1e000)
	libmount.so.1 => /usr/lib/x86_64-linux-gnu/libmount.so.1 (0x00007f574279b000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f57426da000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f5742c62000)
	libpixman-1.so.0 => /usr/lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007f574262e000)
	libxcb-shm.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f5742629000)
[... 10 lines cut ...]
	libatomic.so.1 => /usr/lib/x86_64-linux-gnu/libatomic.so.1 (0x00007f57424e5000)
	libblkid.so.1 => /usr/lib/x86_64-linux-gnu/libblkid.so.1 (0x00007f574248b000)
	libbz2.so.1 => /usr/lib/x86_64-linux-gnu/libbz2.so.1 (0x00007f5742477000)
	libbrotlidec.so.1 => /usr/lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007f574246a000)
[... cut ...]
* libiio.so.1 : [ubuntu 20.04 : version 0.19](https://packages.ubuntu.com/focal/libiio0) ; [ubuntu 22.04 :  version 0.23](https://packages.ubuntu.com/jammy/libiio0), 0.24 available as deb.

I see that libiio package is compiled with all back-ends enabled (local xml ip (with avahi) usb serial), that means we can probably set it as a dependency in the deb package alongside the others libs, fetching from the main repos.
libgtkdatabox 0.9 from ubuntu 20.04 is an issue, could we statically link it? Or is not providing the deb package for 20.04 an option?

gastmaier, 2023-03-23:
Pull request #422 should create the debs for iio-osc.

Some notes:
* I set the minimum libiio version to 0.23, which is the minimum version to compile the master branch (e.g. iio_device_get_label requires >= 0.23) , for the user, that means:
  - Ubuntu 22.04: can install from the pkg manager.
  - Ubuntu 20.04: must install from the GitHub release [1].
* I'm bundling gtkdatabox 1.0.0 (not in Ubuntu 20.04), libad9166-iio, and libad9361-iio **shared libraries** with iio-osc, so the user does not have to install those manually; alternatively:
  - We could statically link them.
  - Create/updated the debs for  them, requiring the user to install them first.

[1] libiio from github install as libiio, while the one from the package manager is named libiio0, 
that's an issue for setting the dependencies, because libiio0 and libiio are treated as different packages, both providing libiio.so.0 and therefore conflicting with each other. Is installing libiio from GitHub the recommended method? If so, can I set the dependency as libiio instead of libiio0, right?

nunojsa, 2023-03-24:
Some notes/thoughts from my side...

> Ubuntu 20.04: must install from the GitHub release [1].

but are you doing anything special for this? I was going to say to just ship an older version of osc with libioo < 0.23 as ubuntu 20.04 is stuck with pretty outdated packages anyways... However, we leaved this far without any package so I would just start to support it from now on. Meaning, that in the current state, we'll just have the pkg in ubuntu 22.04.

> I'm bundling ... 

What do you mean by this? The expectation is that one runs `apt install iio-oscilloscope (or osc)` and everything needed to run the app gets installed. I would test the installer in a ubuntu 22.04 docker container as those come with a minimal set of packages installed. That would be a nice test for you to make sure you're not missing any dependency...

>  libiio from github install as libiio, while the one from the package manager is named libiio0,
that's an issue for setting the dependencies, because libiio0 and libiio are treated as different packages, both providing libiio.so.0 and therefore conflicting with each other. Is installing libiio from GitHub the recommended method? If so, can I set the dependency as libiio instead of libiio0, right?

@pcercuei can you comment/advise in here?

gastmaier, 2023-03-24:
> I'd ship a regular iio-oscilloscope in 22.04 with dependencies on the distribution's libiio0, libad9361-0 and libgtkdatabox1.

Good, Pull request #422 does that now.

> For 20.04, I'd ship a GTK2 version of iio-oscilloscope

I made the branch [ubuntu20.04-v0.14](https://github.com/analogdevicesinc/iio-oscilloscope/tree/ubuntu20.04-v0.14)  for that, the generated deb I uploaded to [v0.14](https://github.com/analogdevicesinc/iio-oscilloscope/releases/tag/v0.14-master) release as adi-osc-ubuntu20.04.deb.

How I tested pull request #422 and ubuntu20.04-v0.14 branch:
- I installed the debs in clean Ubuntu images (20.04 and 22.04) in podman.
- I searched for missing shared libraries with ldd in `bin/osc` and `lib/libosc.so`

We still need to test the app itself, and then we are good to go I believe.

> I assume the .deb meta-data has a way to specify that it needs libiio or libiio0 anyway?

Yes there is, I overlooked the "or |" operator.

---
For any other linux distro/release, we have the #421 flatpak.