analogdevicesinc/iio-oscilloscope #432: error: use of undeclared identifier 'M_PI'

yurivict, 2023-06-03:
```
/usr/ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.16-master/oscplot.c:898:20: error: use of undeclared identifier 'M_PI'
                double a = 2.0 * M_PI / (n - 1);
                                 ^
/usr/ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.16-master/oscplot.c:909:14: error: use of undeclared identifier 'M_PI'
                double a = M_PI * j / (n - 1);
                           ^
```

Version: 0.16
FreeBSD 13.2

gastmaier, 2023-06-03:
M_PI is defined in the the [math.h header](https://www.man7.org/linux/man-pages/man0/math.h.0p.html), check if your system defines them somewhere like `/usr/include/math.h`, and if CMake is including the system headers properly.

yurivict, 2023-06-03:
This looks like a system headers error.

Thank you for your help.

dNechita, 2023-06-06:
Closing this, then.

closed 2023-06-06 by dNechita