analogdevicesinc/iio-oscilloscope #433: error: 'gtk_tool_button_set_stock_id' is deprecated: Use 'gtk_tool_button_set_icon_name' instead [-Werror,-Wdeprecated-declarations]

yurivict, 2023-06-03:
```
/usr/ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.16-master/oscplot.c:6856:10: error: 'gtk_tool_button_set_stock_id' is deprecated: Use 'gtk_tool_button_set_icon_name' instead [-Werror,-Wdeprecated-declarations]
         gtk_tool_button_set_stock_id(GTK_TOOL_BUTTON(priv->fullscreen_button), "gtk-fullscreen");
         ^
/usr/local/include/gtk-3.0/gtk/gtktoolbutton.h:96:1: note: 'gtk_tool_button_set_stock_id' has been explicitly marked deprecated here
GDK_DEPRECATED_IN_3_10_FOR(gtk_tool_button_set_icon_name)
^
```

Version: 0.16
clang-15
FreeBSD 13.2

gastmaier, 2023-06-03:
M_PI is defined in the the [math.h header](https://www.man7.org/linux/man-pages/man0/math.h.0p.html), check if your system defines them somewhere like `/usr/include/math.h`, and if CMake is including the system headers properly.

tfcollins, 2023-06-12:
@yurivict thanks for the bug report. Unfortunately no one really develops on BSD on the team and CI is pretty nonexistent for it. If you can submit a fix that would be great otherwise this will likely be awhile before we can verify and fix on our end.

-Travis

yurivict, 2023-06-18:
This is a GTK-related error.

What version of GTK do you use? Does it build with gtk3-3.24.34 ?

nunojsa, 2023-12-14:
I can also confirm that building for Fedora is pretty broken... We need to either fix it or not treat deprecated stuff as an error (just delaying fixing the real issue but better than nothing for now I guess...).

And I would bet arch linux is likely also affected (will know about it tomorrow)