analogdevicesinc/iio-oscilloscope #437: Libidn issue under Ubuntu 22.04LTS version

danirojas037, 2023-06-28:
Hi everyone,

I've been trying to install IIO Oscilloscope (Ubuntu 22.04 LTS)  following the indications: https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope.
But when I tried to install the libiio library(https://wiki.analog.com/resources/tools-software/linux-software/libiio#how_to_build_it) or any of the other required libraries, after the execution of the "cmake ../" command, I get following error:
 make: error while loading shared libraries: libidn.so.11: cannot open shared object file: No such file or directory

It seems that package libidn11 is not included into 22.04LTS version.
Any solution to fix this?

Regards,

Dani

gastmaier, 2023-06-28:
Hi, try
```
sudo apt install libidn11-dev
```
Installing the development package version is the solution to any missing libraries that you might encounter.
[This Containerfile ](https://github.com/analogdevicesinc/iio-oscilloscope/blob/407a33f83cbb4b5b91b6f0c5353302254f60fec9/.github/workflows/build-deb-ubuntu-22.04.yml) has a full to compile with Ubuntu 22.04LTS.

danirojas037, 2023-06-28:
> 

This works, thanks!

Regards,

Dani