analogdevicesinc/iio-oscilloscope #438: ZCU102+FMCOMMS2 not detected by IIO Oscilloscope

danirojas037, 2023-06-28:
Hi,

I'm running a no-OS project with the following setup: ZCU102+FMCOMMS2, and trying to observe the spectrum of a received signal with an IIO Oscilloscope installed on a host (my desktop computer) and connected to the setup through serial connection, but the IIO Oscilloscope it is not detecting the board:
[image]

Any possible solution? (I've done a test with the PlutoSDR, and it is detected by the IIO Oscilloscope)

Thanks,

Dani

tfcollins, 2023-06-28:
Have you verified that the libiio context has started on the board itself and nothing else is blocking the UART interface?

danirojas037, 2023-06-28:
> Have you verified that the libiio context has started on the board itself and nothing else is blocking the UART interface?
How can I verify that the libiio context has started on the board?

tfcollins, 2023-06-28:
UART on the board will provide feedback on boot that you can connect to the device over libiio. It should give you the required baudrate. What appears on your system at boot?

Beauxrel, 2023-06-29:
Double check your .xdc file. I noticed that I had defined the wrong fmc port. Double check with you user guide on which is which

danirojas037, 2023-07-06:
> 

I was able to solve it! 
The problem was that I haven't applied the configuration that appears on my system at boot:
[image]

When I applied this configuration to the IIO Oscilloscope, it detects the board:
[image]

Thanks for the help,

Dani

closed 2023-07-06 by tfcollins