analogdevicesinc/iio-oscilloscope #439: VCU108 + FMCOMMS3 not recognized by iio-oscilliscope

Beauxrel, 2023-06-29:
Ramdisk addr 0x00000000,
Compiled-in FDT at 0x80649d44
earlycon: uartlite_a0 at MMIO 0x40600000 (options '115200n8')
printk: bootconsole [uartlite_a0] enabled
cma: Reserved 512 MiB at 0x8e000000
Linux version 5.10.0-98747-g255f8a9b983b (jenkins@romlxbuild1) (microblazeel-xilinx-linux-gcc.real (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.1) #1920 Fri Apr 14 08:25:59 EEST 2023
setup_memory: max_mapnr: 0x7ffff
setup_memory: min_low_pfn: 0x80000
setup_memory: max_low_pfn: 0xb0000
setup_memory: max_pfn: 0xfffff
Zone ranges:
  DMA      [mem 0x0000000080000000-0x00000000afffffff]
  Normal   empty
  HighMem  [mem 0x00000000b0000000-0x00000000ffffefff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000080000000-0x00000000ffffefff]
Initmem setup node 0 [mem 0x0000000080000000-0x00000000ffffefff]
setup_cpuinfo: initialising cpu 0
setup_cpuinfo: Using full CPU PVR support
wt_msr_noirq
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists, mobility grouping on.  Total pages: 522751
Kernel command line: earlycon
Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
mem auto-init: stack:off, heap alloc:off, heap free:off
Memory: 1536816K/2097148K available (6439K kernel code, 597K rwdata, 5704K rodata, 200K init, 178K bss, 36044K reserved, 524288K cma-reserved, 1310716K highmem)
NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
irq-xilinx: mismatch in kind-of-intr param
irq-xilinx: /amba_pl/interrupt-controller@41200000: num_irq=16, sw_irq=0, edge=0xffffc5f0
xilinx_timer_init: Timer base: 0xb0002000, Clocksource base: 0xb0002010
clocksource: xilinx_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604467 ns
sched_clock: 32 bits at 100MHz, resolution 10ns, wraps every 21474836475ns
/amba_pl/timer@41c00000: irq=1, cpu_id 0
xilinx_timer_shutdown
xilinx_timer_set_periodic
Calibrating delay loop... 49.35 BogoMIPS (lpj=246784)
pid_max: default: 4096 minimum: 301
Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
devtmpfs: initialized
random: get_random_u32 called from bucket_table_alloc.isra.0+0x148/0x170 with crng_init=0
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 16 (order: -4, 448 bytes, linear)
NET: Registered protocol family 16
DMA: preallocated 256 KiB GFP_KERNEL pool for atomic allocations
DMA: preallocated 256 KiB GFP_KERNEL|GFP_DMA pool for atomic allocations
PCI: Probing PCI hardware
vgaarb: loaded
jesd204: found 0 devices and 0 topologies
clocksource: Switched to clocksource xilinx_clocksource
NET: Registered protocol family 2
tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 12288 bytes, linear)
TCP established hash table entries: 8192 (order: 3, 32768 bytes, linear)
TCP bind hash table entries: 8192 (order: 5, 163840 bytes, linear)
TCP: Hash tables configured (established 8192 bind 8192)
UDP hash table entries: 512 (order: 2, 24576 bytes, linear)
UDP-Lite hash table entries: 512 (order: 2, 24576 bytes, linear)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
PCI: CLS 0 bytes, default 32
random: fast init done
workingset: timestamp_bits=30 max_order=19 bucket_order=0
Key type cifs.idmap registered
jffs2: version 2.2. (NAND) (SUMMARY)  © 2001-2006 Red Hat, Inc.
[... 640 lines cut ...]
                                attr  4: samples_pps ERROR: No such device (19)
                                attr  5: sampling_frequency value: 30720000
                        voltage3:  (input, index: 3, format: le:S12/16>>0)
                        6 channel-specific attributes found:
                                attr  0: calibbias value: 0
                                attr  1: calibphase value: 0.000000
                                attr  2: calibscale value: 1.000000
                                attr  3: label ERROR: Function not implemented (38)
                                attr  4: samples_pps ERROR: No such device (19)
                                attr  5: sampling_frequency value: 30720000
                2 device-specific attributes found:
                                attr  0: sync_start_enable value: disarm
                                attr  1: sync_start_enable_available value: arm
                3 buffer-specific attributes found:
                                attr  0: data_available value: 0
                                attr  1: length_align_bytes value: 8
                                attr  2: watermark value: 2048
                2 debug attributes found:
                                debug attr  0: pseudorandom_err_check value: CH0 : PN9 : Out of Sync : PN Error
CH1 : PN9 : Out of Sync : PN Error
CH2 : PN9 : Out of Sync : PN Error
CH3 : PN9 : Out of Sync : PN Error
                                debug attr  1: direct_reg_access value: 0x0
                No trigger on this device
#

Beauxrel, 2023-06-29:
I know that's a lot, but the beginning is the startup and then I called iio_info.

For some reason iio_osc is not recognizing it.

On refresh I either get 

Could not get IIO Context: Input/output error... [Happens when I leave putty open]

or 

Could not get IIO Context: Invalid argument... [Happens after I close putty]

tfcollins, 2023-06-29:
Are you able to ping and SSH to the board from the host PC running IIO-Scope?

Beauxrel, 2023-06-29:
I cannot ssh into the board.
192.168.2.1 works for my plutosdr

but the ip that is on the vcu108 which i believe is also 192.168.2.1, does not allow me to connect, I can connect with serial though settings: COM4, 115200

[image]

Beauxrel, 2023-06-29:
Is internet access required for both the host and target?

Beauxrel, 2023-06-29:
We are able to ping and ssh into the device.

Beauxrel, 2023-06-29:
I was able to get the iio-oscilliscope to work.

Solution
---------
device (vcu108) needed to be on the network
uri = ip:xx.xx.xx.xx
connect :D 

test
-----
ping and ssh inet

Beauxrel, 2023-06-30:
@tfcollins is this something I should be concerned about?

`at24` 3-0050: supply vcc not found, using dummy regulator`

tfcollins, 2023-06-30:
No. It can be ignored

closed 2023-07-01 by Beauxrel