analogdevicesinc/iio-oscilloscope #441: iio-oscilloscope is broken: it uses deprecated functions: gdk_screen_get_height, gtk_tool_button_set_stock_id, gtk_misc_set_alignment

yurivict, 2023-07-11:
gtk3-3.24.34 has several function used in iio-oscilloscope deprecated:
```
/usr/ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.16-master/osc.c:475:18: error: 'gdk_screen_get_height' is deprecated [-Werror,-Wdeprecated-declarations]
        gint screen_h = gdk_screen_get_height(screen);
                        ^
/usr/local/include/gtk-3.0/gdk/gdkscreen.h:59:1: note: 'gdk_screen_get_height' has been explicitly marked deprecated here
GDK_DEPRECATED_IN_3_22
^
/usr/local/include/gtk-3.0/gdk/gdkversionmacros.h:412:47: note: expanded from macro 'GDK_DEPRECATED_IN_3_22'
# define GDK_DEPRECATED_IN_3_22               GDK_DEPRECATED
                                              ^
/usr/local/include/gtk-3.0/gdk/gdkversionmacros.h:48:24: note: expanded from macro 'GDK_DEPRECATED'
#define GDK_DEPRECATED G_DEPRECATED _GDK_EXTERN
                       ^
/usr/local/include/glib-2.0/glib/gmacros.h:1262:37: note: expanded from macro 'G_DEPRECATED'
#define G_DEPRECATED __attribute__((__deprecated__))
                                    ^
```

```
/usr/ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.16-master/plugins/dac_data_manager.c:904:3: error: 'gtk_misc_set_alignment' is deprecated [-Werror,-Wdeprecated-declarations]
                gtk_misc_set_alignment(GTK_MISC(scale), 0.0, 0.5);
                ^
/usr/local/include/gtk-3.0/gtk/deprecated/gtkmisc.h:71:1: note: 'gtk_misc_set_alignment' has been explicitly marked deprecated here
GDK_DEPRECATED_IN_3_14
^
/usr/local/include/gtk-3.0/gdk/gdkversionmacros.h:356:47: note: expanded from macro 'GDK_DEPRECATED_IN_3_14'
# define GDK_DEPRECATED_IN_3_14               GDK_DEPRECATED
                                              ^
/usr/local/include/gtk-3.0/gdk/gdkversionmacros.h:48:24: note: expanded from macro 'GDK_DEPRECATED'
#define GDK_DEPRECATED G_DEPRECATED _GDK_EXTERN
                       ^
/usr/local/include/glib-2.0/glib/gmacros.h:1262:37: note: expanded from macro 'G_DEPRECATED'
#define G_DEPRECATED __attribute__((__deprecated__))
                                    ^
```

Please share the GTK version that you use.

This was previously reported [here](https://github.com/analogdevicesinc/iio-oscilloscope/issues/433) but didn't receive sufficient attention there.

Version: 0.16

yurivict, 2023-12-07:
@cristina-suteu @andrei47w 

Hello,

Do you know why the released code uses deprecated GDK functions?
Several last releases, including 0.17, fail to build on systems with the latest GTK.

The dependencies that we use are:
```
===>   iio-oscilloscope-0.14_4 depends on file: /usr/local/sbin/pkg - found
===>   iio-oscilloscope-0.14_4 depends on shared library: libad9361.so - found (/usr/local/lib/libad9361.so was installed by package libad9361-iio-0.3)
===>   iio-oscilloscope-0.14_4 depends on shared library: libcurl.so - found (/usr/local/lib/libcurl.so was installed by package curl-8.4.0)
===>   iio-oscilloscope-0.14_4 depends on shared library: libfftw3.so - found (/usr/local/lib/libfftw3.so was installed by package fftw3-3.3.10_4)
===>   iio-oscilloscope-0.14_4 depends on shared library: libfontconfig.so - found (/usr/local/lib/libfontconfig.so was installed by package fontconfig-2.14.2,1)
===>   iio-oscilloscope-0.14_4 depends on shared library: libfreetype.so - found (/usr/local/lib/libfreetype.so was installed by package freetype2-2.13.1)
===>   iio-oscilloscope-0.14_4 depends on shared library: libgtkdatabox.so - found (/usr/local/lib/libgtkdatabox.so was installed by package gtkdatabox-1.0.0)
===>   iio-oscilloscope-0.14_4 depends on shared library: libiio.so - found (/usr/local/lib/libiio.so was installed by package libiio-0.25)
===>   iio-oscilloscope-0.14_4 depends on shared library: libjansson.so - found (/usr/local/lib/libjansson.so was installed by package jansson-2.14)
===>   iio-oscilloscope-0.14_4 depends on shared library: libmatio.so - found (/usr/local/lib/libmatio.so was installed by package matio-1.5.26)
===>   iio-oscilloscope-0.14_4 depends on shared library: libatk-1.0.so - found (/usr/local/lib/libatk-1.0.so was installed by package at-spi2-core-2.50.0)
===>   iio-oscilloscope-0.14_4 depends on shared library: libcairo.so - found (/usr/local/lib/libcairo.so was installed by package cairo-1.17.4_2,3)
===>   iio-oscilloscope-0.14_4 depends on shared library: libgdk_pixbuf-2.0.so - found (/usr/local/lib/libgdk_pixbuf-2.0.so was installed by package gdk-pixbuf2-2.42.10_1)
===>   iio-oscilloscope-0.14_4 depends on shared library: libglib-2.0.so - found (/usr/local/lib/libglib-2.0.so was installed by package glib-2.78.0,2)
===>   iio-oscilloscope-0.14_4 depends on shared library: libintl.so - found (/usr/local/lib/libintl.so was installed by package gettext-runtime-0.22.3)
===>   iio-oscilloscope-0.14_4 depends on shared library: libgtk-x11-2.0.so - found (/usr/local/lib/libgtk-x11-2.0.so was installed by package gtk2-2.24.33_1)
===>   iio-oscilloscope-0.14_4 depends on shared library: libxml2.so - found (/usr/local/lib/libxml2.so was installed by package libxml2-2.10.4_1)
===>   iio-oscilloscope-0.14_4 depends on shared library: libpango-1.0.so - found (/usr/local/lib/libpango-1.0.so was installed by package pango-1.50.14)
```

andrei47w, 2023-12-08:
Hi,

The code is not yet updated for newer versions of gtk since we are still using gtk+ 3.0, version 3.24.36.

nunojsa, 2023-12-14:
@andrei47w ,

I think this becoming an important thing to handle... Fedora38 is also broken and I bet arch linux is likely the same

nunojsa, 2023-12-14:
I mean, this is not a proper fix but at least in the meantime we could do this:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 0650792..20aef00 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -93,6 +93,7 @@ add_compile_options(-Wall -Wextra -Werror
        -Wno-unused-parameter
        -Wno-error=unused-but-set-variable
        -Wno-error=unused-function
+       -Wno-error=deprecated-declarations
        -Wimplicit-fallthrough
        -Wvla
        -std=gnu90
```

I would prefer for someone to fix the real problem. Unfortunately I can't do it myself right now (too many things going on) but I can very well send the above PULL if no else is available now to fix this.

@dNechita , do you think there's someone who could tackle this?

cristina-suteu, 2023-12-14:
@nunojsa I will start working on the proper fix next week , in the mean time if you want to create a PR with the addendum that it’s only a temporary fix, I think it would be great

nunojsa, 2023-12-14:
> @nunojsa I will start working on the proper fix next week , in the mean time if you want to create a PR with the addendum that it’s only a temporary fix, I think it would be great

Since you're looking at it next week it might not be worth it... Should not be that bad to fix this. But if it turns out to be a bigger effort than expected, ping me and I'll send the PULL.

dNechita, 2023-12-14:
> Adding `-Wno-error=deprecated-declarations` isn't sufficient to unbreak it.
> 
> There are additional errors:
> 
> ```
> In file included from /usr/ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.14-master/datatypes.c:9:
> In file included from /usr/ports/misc/iio-oscilloscope/work/iio-oscilloscope-0.14-master/datatypes.h:12:
> In file included from /usr/local/include/gtkdatabox/gtkdatabox.h:24:
> /usr/local/include/gtkdatabox/gtkdatabox_graph.h:100:10: error: unknown type name 'GdkRGBA'
>                                      GdkRGBA * color);
>                                      ^
> /usr/local/include/gtkdatabox/gtkdatabox_graph.h:101:4: error: unknown type name 'GdkRGBA'
>    GdkRGBA *gtk_databox_graph_get_color (GtkDataboxGraph * graph);
>    ^
> /usr/local/include/gtkdatabox/gtkdatabox_graph.h:104:10: error: unknown type name 'GdkRGBA'
>                                      GdkRGBA * rgba);
>                                      ^
> /usr/local/include/gtkdatabox/gtkdatabox_graph.h:105:4: error: unknown type name 'GdkRGBA'
>    GdkRGBA *gtk_databox_graph_get_rgba (GtkDataboxGraph * graph);
>    ^
> 4 errors generated.
> ```
> 
> iio-oscilloscope-0.14 clang-16 FreeBSD 13.2

Hi,
These errors about gtkdatabox have nothing to do with the original issue (osc using some deprecated gtk functions). So the fix provided by Nuno (to make the compiler not treat the usage of deprecated functions as errors) can be very useful. Using deprecated functions isn't the worse thing. Functions could have been marked obsolete or even removed entirely. But since we set the compiler to consider the usage or deprecated functions an error, then it's our fault for sure and we didn't do it based on some strategy to enforce the usage of the latest API GTK has to offer. We just added -Werror to treat all warnings as errors as an initiative to improve code quality but we didn't foresee the magnitude of the maintenance.

Getting back to the gtkdatabox issue, it looks like your build is trying to use gtkdatabox for GTK3 (version 1.0.0) but iio-oscilloscope v0.14 is based on GTK2. So, your build needs to use gtkdatabox for GTK2 (latest version seems to be 0.9.3.1). The iio-oscilloscope version 0.15 is the latest release that uses GTK2. The next release (v0.16) is already using GTK3.

nunojsa, 2023-12-15:
> Getting back to the gtkdatabox issue, it looks like your build is trying to use gtkdatabox for GTK3 (version 1.0.0) but iio-oscilloscope v0.14 is based on GTK2. So, your build needs to use [...]

Yeah, I was going to suggest moving to v0.17 because I was not seeing those issues. I mean, I'm also compiling with clang so there are still some minor issues that I will fix (mainly unused variables) but I was able to compile ignoring the warning.

> We just added -Werror to treat all warnings as errors as an initiative to improve code quality but we didn't foresee the magnitude of the maintenance.

I still think it's good to have it. But yeah, along the way we might have to fine tune some warning to not be treated as errors as sometimes compilers also give you false positives...

> Using deprecated functions isn't the worse thing. Functions could have been marked obsolete or even removed entirely.

Indeed but in this case we should fix it otherwise we'll just get into the point where those functions are removed and we can't compile anymore. I mean, I'm assuming they will be removed otherwise I'm not seeing the point of this deprecated bloat... Personally, I think doing this kind of crap is horrible. Just implement your new features and APIs , document that the old ones should not be used anymore but leave them (and the functionality) there. Breaking your users like this is just not cool. Anyways, don't want to start an argument... just wanted to take it out of my system :)

cristina-suteu, 2023-12-15:
@yurivict If you are interested in just running IIO-Oscilloscope on x86_64 type architecture, and do not want to write your own custom plugins, you can download our AppImage for linux, from the latest release, until we fix this is problem. If you got any issues, please report them back to us.

yurivict, 2023-12-15:
@cristina-suteu No, I am interested in fixing the FreeBSD port since I am its maintainer.

nunojsa, 2023-12-19:
Also to mention that this only looks to pop up when using clang...

yurivict, 2024-06-23:
Why don't you fix deprecation issues?

cristina-suteu, 2025-03-28:
deprecation issues finally addressed entirely

closed 2025-03-28 by cristina-suteu