analogdevicesinc/iio-oscilloscope #443: problem in capture

ali127124, 2023-08-07:
I want to connect the ADIS16465-1BMLZ module to the Raspberry Pi 4 using the EVAL-ADISIMU1-RPIZ board. To do this, I installed the Kuiper Linux operating system on the Raspberry Pi and ran the ADI IIO Oscilloscope software. Initially, nothing is displayed. By adding the following lines to the config.txt file as shown below, the data is displayed:

However, nothing is displayed in the plot section. What should I do?"

nunojsa, 2023-08-07:
Hi,

> By adding the following lines to the config.txt file as shown below

Which line?

> However, nothing is displayed in the plot section. What should I do?"

Are you using FFT? Did you already tried to lower the fft size or the number of samples? Also, what channels are you trying to sample? We recently fixed a bug in libiio when you try to sample channels of different sizes. That will happen in your case if you mix temperature with the accel or gyro...

ali127124, 2023-08-07:
[image]

ali127124, 2023-08-07:
no i use real time
[image]
[image]
i have data in dmm but in plot nothing
[image]
[image]

nunojsa, 2023-08-07:
Oh sure... You should not ignore messages " ... needs an impulse generator" :) 

The capture is not even enabled so there's no way to see a thing. You need to right click on the device in the plot channel and then look for the impluse generator setting. IIRC, you'll only have one option so there's no way to have it wrong. Let me know if it works...

ali127124, 2023-08-07:
yes i saw the message but I don't know from which part it can be done, there is nothing in the settings part

ali127124, 2023-08-07:
i right clicked on the device in the plot channel but nothing happened

nunojsa, 2023-08-07:
> i right clicked on the device in the plot channel but nothing happened

Are you sure you're clicking on the right place?

[image]

In the above example, I  right click on top of axi-adrv9002-rx-lpc or in the rx2 device and I can see a pop up with "Trigger settings" and "Impulse generator settings". In my case, there's no impulse generator so I can't really click it.

ali127124, 2023-08-07:
wow thaks so much yes it works

ali127124, 2023-08-07:
can i save the data of the plot in .csv file?

ali127124, 2023-08-07:
Is it possible to save information for a period of time, for example 5 minutes?

ali127124, 2023-08-07:
> Are you sure you're clicking on the right place?
>
> [image]
>
> In the above example, I right click on top of axi-adrv9002-rx-lpc or in the rx2 device and I can see a pop up with "Trigger settings" and [...]
wow thaks so much yes it works .Is it possible to save information for a period of time, for example 5 minutes?

nunojsa, 2023-08-07:
> wow thaks so much yes it works .Is it possible to save information for a period of time, for example 5 minutes?

Not that I'm aware off. But maybe @dNechita can give you confirmation on that...

ali127124, 2023-08-07:
> > wow thaks so much yes it works .Is it possible to save information for a period of time, for example 5 minutes?
> 
> Not that I'm aware off. But maybe @dNechita can give you confirmation on that...

thanks so much for your help

tfcollins, 2023-08-08:
@ali127124  it isn't really possible to do a 5 minute capture. You would need to go to C or Python

ali127124, 2023-08-09:
> @ali127124 it isn't really possible to do a 5 minute capture. You would need to go to C or Python

do you know any python library or python example code that I can use for my purpose?

tfcollins, 2023-08-15:
@ali127124  sorry missed the last question. So you can use pylibiio directly or modify an existing IMU class for ADIS16465.
- pylibiio: http://analogdevicesinc.github.io/libiio/v0.25/python/index.html
- pyadi-iio IMU class example: https://github.com/analogdevicesinc/pyadi-iio/blob/master/adi/adis16460.py

ali127124, 2023-08-15:
> @ali127124 sorry missed the last question. So you can use pylibiio directly or modify an existing IMU class for ADIS16465.
>
> * pylibiio: [...]

thanks much for getting back to me, my problem solved and I was forced to make a little change on pylibiio library(adis16495) in order to could connect to my sensor and got information from my sensor(adis16465)
anyway thanks for your help

tfcollins, 2023-08-15:
Great! Looks like your issue is solved. I'll close the issue for now then.

closed 2023-08-15 by tfcollins