analogdevicesinc/iio-oscilloscope #461: Installation error

NIDHIPANDA, 2023-10-29:
Hello, 

I am trying to install IIO on ubuntu 20.04 system. On following the installation steps on "https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope#installation", while installing libiio using "https://wiki.analog.com/resources/tools-software/linux-software/libiio#how_to_build_it" link gives following error. 

/libiio/build$ cmake ../ && make && sudo make install
-- cmake version: 3.16.3
CMake Error at CMakeLists.txt:428 (message):
  Unable to find libzstd dependency.

  If you want to disable ZSTD compression support, set WITH_ZSTD=OFF.

-- Features enabled : xml zstd network dns-sd avahi ipv6 local local-dmabuf local-mmap hwmon usb utils iiod udev-rule iiod-serial iiod-aio iiod-usb man-utils
-- Features disabled: bonjour serial examples modules usb-dynamic network-dynamic serial-dynamic iiod-systemd iiod-sysvinit iiod-upstart doc man python-bindings c#-bindings c++-bindings
-- LOG_LEVEL set to "Info"
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/cipl/libiio/LIBZSTD_INCLUDE_DIR
   used as include directory in directory /home/cipl/libiio
/home/cipl/libiio/iiod/LIBZSTD_INCLUDE_DIR
   used as include directory in directory /home/cipl/libiio/iiod
LIBZSTD_LIBRARIES
    linked by target "iio" in directory /home/cipl/libiio
    linked by target "iiod" in directory /home/cipl/libiio/iiod

-- Configuring incomplete, errors occurred!
See also "/home/cipl/libiio/build/CMakeFiles/CMakeOutput.log".

 I tried by using "set WITH_ZSTD=OFF" but getting same results. 
How to pass this error.

mhennerich, 2023-10-30:
Have you tried ?
#sudo apt-get install libzstd-dev

>I tried by using "set WITH_ZSTD=OFF" but getting same results.

Works for me...

```
libiio$ cmake -DWITH_ZSTD=OFF .
-- cmake version: 3.22.1
-- Looking for libserialport : Found
-- bin= lib= inc=
-- Python_EXECUTABLE /usr/bin/python3.10
-- Features enabled : xml network dns-sd avahi ipv6 serial local local-mmap usb utils iiod udev-rule iiod-serial iiod-aio iiod-usb man-utils python-bindings
-- Features disabled: zstd bonjour hwmon examples c++-bindings iiod-systemd iiod-sysvinit iiod-upstart doc man c#-bindings
-- LOG_LEVEL set to "Info"
-- Configuring done
-- Generating done
-- Build files have been written to: /home/michael/devel/git/libiio
```

```
libiio$ cmake -DWITH_ZSTD=ON .
-- cmake version: 3.22.1
-- Looking for libserialport : Found
-- bin= lib= inc=
-- Python_EXECUTABLE /usr/bin/python3.10
-- Features enabled : xml zstd network dns-sd avahi ipv6 serial local local-mmap usb utils iiod udev-rule iiod-serial iiod-aio iiod-usb man-utils python-bindings
-- Features disabled: bonjour hwmon examples c++-bindings iiod-systemd iiod-sysvinit iiod-upstart doc man c#-bindings
-- LOG_LEVEL set to "Info"
-- Configuring done
-- Generating done
-- Build files have been written to: /home/michael/devel/git/libiio
```

NIDHIPANDA, 2023-10-30:
Thank you for your reply. After installing libzstd-dev now the libiio is installed. However same error is there on the installation of the libad9361-iio libraries .

/libad9361-iio$ cmake ./
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen dot 
-- Using default dependencies for packaging
-- Package dependencies: libc6-dev (>= 2.19)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBIIO_INCLUDEDIR
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/test
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings
   used as include directory in directory /home/cipl/libad9361-iio/bindings

Can you please help me with this.

adisuciu, 2023-10-31:
You need libiio-v0 branch. The build instructions on the wiki were outdated. I updated them.
Go to the libiio build folder
git checkout libiio-v0
rebuild libiio and then rebuild iio-osc

NIDHIPANDA, 2023-10-31:
Thank you  for your reply. Now, the installation is  complete without any error. I am also able to see IIO oscilloscope in applications, but when I launch it it asks for authentication and after giving permission it does not launch. Can you please suggest what went wrong.

NIDHIPANDA, 2023-11-01:
Thank you. It worked.

closed 2023-11-01 by NIDHIPANDA