analogdevicesinc/iio-oscilloscope #489: Plot limited to sampling frequency

javiercarrascocruz, 2024-04-14:
Hi, I am connected to a remote target over the network and I am able to read properties and single values from an IIO device (hts221), which provides a triggered buffer, and its maximum sampling frequency is 13 samples/second.

The plot view only lets me capture up to 13 samples in both single shot and continuous measurement. If I try to read more samples than the sampling frequency (e.g. 14, any value greater than 13), the application freezes and I see on the target that iiod throws a timeout:

iiod[541]: WARNING: High-speed mode not enabled
iiod[541]: ERROR: Reading from device failed: -110

Is the plot view limited to the samples than can be taken in one second?

When at it, is there any way to get processed samples (offset and scale applied) in the plot view like it is already done in the DMM mode? Thanks.

dNechita, 2024-06-03:
Hi,
The plot view is not expected to enforce such a limit. Error 110 means that connections has timed out. What iio-osciloscope has, is an internal hardcoded timeout. One way to go forward would be to modify this internal value which can be done via the .ini file. Also make sure you have the latest iio-oscillscope (the latest from main branch and not the latest release).
For example, to force the internal timeout value to 5 seconds, you can modify the .ini file (that gets saved once the iio-osc closes and gets loaded when the iio-osc opens up) like this:
"capture_timeout=5000"

Regarding applying offset and scale, yes, the plot allows adding some simple math operations for each channel. Just right click on a channel and select the 'math' option. Then you will be able to provide an offset to be added to the signal and a gain value to be multiplied to the signal.

I apologize for the late reply.

dNechita, 2024-12-17:
@dlech, you are correct that iio-oscilloscope does not currently support multi-channel devices that are self-triggered. When the timeout computation code was written, it did not account for the use case you mentioned. This appears to be an improvement that should be added to the iio-osc requirements list.
For clarity, for anyone reading this thread, iio-oscilloscope does include a mechanism to override the computed value based on the sampling frequency attribute. The override value is read from the .ini file.
https://github.com/analogdevicesinc/iio-oscilloscope/blob/fbd5aa5b16d839e8142705fd22881090dd3d9129/osc.c#L1519-L1524