analogdevicesinc/iio-oscilloscope #508: Trojan - IIO-Oscilloscope - v0.17-master

Pekiel, 2024-12-06:
Hello,

IIO-Oscilloscope - v0.17-master for Windows is detected as Trojan by ESET LiveGuard.
Palo Alto is detecting file as "Virus/Linux.WGeneric.efjyuh" by WildFire.

Pekiel, 2024-12-06:
Yes. Directly from this site. When I download the file, ESET is imidietly deleting the file.
[image]

Pekiel, 2024-12-06:
My Firewall (Palo Alto) have the problem with the file for Linux
[image]

dNechita, 2024-12-17:
Thanks for reporting this!
All installers available at: https://github.com/analogdevicesinc/iio-oscilloscope/releases are digitally signed, which should typically reassure antivirus software that the installer is not a threat. However, it seems this is not sufficient in this case. I will investigate further to determine what additional actions are needed to prevent future false positive detections.
If your antivirus is blocking the download, you can add the installer to the whitelist.